PhilipsHue/flutter_reactive_ble #137: ios subscribeToCharacteristic crashes the app

dawidb-appdate, 2020-11-04:
**Describe the bug**
Subsrcibing to specific characteristic crashes ios app immediately. Android one works well with the same code.

**To Reproduce**
Steps to reproduce the behavior:
1. use sample app 
2. change body of the function _Future<void> discoverServices(String deviceId) async_ 
to something like 
`final characteristic = QualifiedCharacteristic(serviceId: Uuid.parse('4c0719b0-c810-492b-87b1-63c6c3fd584f'), characteristicId: Uuid.parse('4c0719b3-c810-492b-87b1-63c6c3fd584f'), deviceId: deviceId);
    
      _ble.subscribeToCharacteristic(characteristic).listen((data) {
        print(data.toString());
      }, onError: (dynamic error) {
        print(error);
      });`
3. launch the app
4. tap search button
5. select device
6. tap connect and wait until status changes
7. tap Discover services
8. app immediately crashes - tries to subsrcibe to characteristic

Note: original code for discovering services works properly on both ios and android, so service and characteristics are discovered
.

**Expected behavior**
Should receive data printed on console - Android works properly

**Smartphone / tablet**
 - Device:iPhone XS
 - OS: iOS 14
 - Package version: 2.6.1+1

Attached screenshot from xcode
[image]

werediver, 2020-11-05:
Thank you for reporting this.

I have an idea why this can happen and how the issue can be fixed (mitigated).

dawidb-appdate, 2020-11-09:
hey @werediver, have you by any chance figure this one out? thnks :)

werediver, 2020-11-09:
@danielroek Yes, the pull-request #139 should fix this issue. We now have issues with CI, this is why the pull-request is not merged yet. As of November 1st Codemagic stopped providing "teams" functionality for free, which breaks our integration.

You should be able to use the corresponding branch from this repository via [dependency_override](https://dart.dev/tools/pub/pubspec#dependencies) option in your top-level package (the app). If you do that, please let us know whether the fix works for you.

danielroek, 2020-11-09:
I believe @dawidb-appdate was supposed to be mentioned

dawidb-appdate, 2020-11-12:
@werediver it seems to fix the issue 
just fyi  it's crashing in debug scheme (release seems to be working), so you're unable to work with live data and console on for iOS devices. if you build an .ipa for ios (adhoc/appstore) it will work properly. as a workaround - android is working in debug mode
will it be tagged on master/released anytime soon?

werediver, 2020-11-12:
@dawidb-appdate It could have worked in release mode without the fix anyway. Does the fix actually fix anything for you? Did you manage to override your dependency with the version from the branch (the feature-branch originally, but now "master", because the pull-request was integrated)? A release is planned in soon, perhaps after #141 is merged.

dawidb-appdate, 2020-11-12:
i'll check 2.6.1+1 in the release mode to confirm if it really was only debug issue.
for now, master version is still crashing in debug mode on iOS (per description)

dawidb-appdate, 2020-11-17:
sorry it took so long, here are my test results:
2.7.1, 2.6.1+1 and 2.5.3 has the same issue: release app works properly, debug app crashes as described 
@werediver

werediver, 2020-11-18:
Thank you for doing the extra tests. Let's reopen then.

Let's treat #142 as a duplicate, as this ticket is older.

werediver, 2020-11-18:
@dawidb-appdate @kapiios To double-check: does this issue happens reliably _every_ time a characteristic is being subscribed to (100% reproducible)?

dawidb-appdate, 2020-11-18:
@werediver yes 100% reproducible in debug mode. Once device start sending data it crashes

kapiios, 2020-11-18:
It is same for me but I'm not really tested it on the release mode only debug.

plyrek, 2020-12-07:
@werediver Has there been any progress on this problem. I can confirm that this is a problem building from vsCode as well. I am sorry for the attachment spam but I am including several screen shots. There is the initial error

[image]

 as well as screen caps of the proceeding 3 or 4 entries in the stack trace. 
[image]
[image]
[image]
[image]

I have also included screen caps after stepping through the initial error. 

[image]

Please let me know if there is anything else I can do for you.

werediver, 2020-12-10:
I will try to get some time for this issue next week (Dec 14th onwards), guys. We do not observe this issue in Philips Hue Bluetooth app, which only makes it weirder.

plyrek, 2020-12-15:
@werediver Thank you for taking some time to look at this issue. We have a client review that has been impacted by this problem. I have a couple of thoughts on the subject to pass along.

Below is a list of the other packages I have installed in my project.
cupertino_icons: ^0.1.3
  firebase_core: ^0.5.2
  firebase_auth: ^0.18.3
  cloud_firestore: ^0.14.3
  provider: ^4.3.2+2
  font_awesome_flutter: ^8.8.1
  rxdart: ^0.24.1
  flutter_reactive_ble: ^2.7.1
  video_player: ^1.0.1
  flutter_animation_progress_bar: ^1.0.6
  modal_progress_hud: ^0.1.3
  
I am wondering if that the problem is a result of a conflict between some package and react_ble. I previously ran into some problems with the Firebase plugins causing some problems, particularly around the Protobuff plugin. I know your team has recently update that plugin to fix the compatibility issue. This is total speculation on my part, but I did notice in the stacktrace I posted above that several steps below the actual breaking error there is a call to Protobuff function. I am wondering if this could be part of the problem? I have included the specific screen shot I am referencing below. Please let me know if there is anything I can do to help further and thank you.
  
  
[image]

plyrek, 2020-12-29:
@werediver Has there been any progress on this issue or any theories on why this might be happening?

werediver, 2020-12-29:
This issue is suck till the next year. I will probably be able to allocate time to it closer to the middle of January.

If you're looking for a workaround, try introducing small delays (100-300 ms) somewhere near the calls causing the issue. If this helps, it's only good as a temporary fix, though.

plyrek, 2020-12-29:
@werediver Thank you for the quick update. My team will be taking a much closer look at this sometime this week. We will really dive into the code and stack trace and also try isolating the package from other packages such as Firebase. I will let you know if we find anything. We will give your suggestion a shot too.

plyrek, 2021-01-14:
@werediver Hello, just checking in. I noticed another thread also mentions subscribingToCharacteristics in the forum but the discussion in that thread is a little over my head technically. Do you think it could have anything to do with the bug we are seeing in iOS. I believe that thread was for Android right?

lwinged, 2021-01-28:
Hello @werediver @remonh87 ,

I think, I have the same issue on iOS when I power off the Bluetooth after subscribed to characteristics.

I subscribe to characteristics when device is connected and when I power off the Bluetooth I have a crash.

Package version: 2.7.2 and 2.4.0+1
iOS 13.4.1
iPhone 6s Plus

[image]
[image]
[image]
[image]

werediver, 2021-02-09:
@lwinged I edited your comment to add collapsible sections, because it was so long. Sorry for intervention. I believe, later you opened a separate ticket #205 for your issue. That is a was a good move, because your case is different (and doesn't seem to be an issue at all).

PR #139 (2020-11-10): Initialize event channels early
Fixes #137

Also corrects an error message mentioned in #138. @HelloBytes Thank you for uncovering it!
closed 2021-10-01 by remonh87