PhilipsHue/flutter_reactive_ble #138: ios readCharacteristic fails

HelloBytes, 2020-11-05:
After connecting to the device I try to read a characteristic. It gives me this error:

flutter: Exception: GenericFailure<CharacteristicValueUpdateError>(code: CharacteristicValueUpdateError.unknown, message: "A service 4C57CE91-D88F-3365-6E2B-D86615BF7C37 is not found in the peripheral 55001000-ABCD-429F-9150-4BF2A5ED8308 (make sure it has been discovered)")

On android it works fine.

The exception is a little bit strange, because its says A **service** 4C57CE91-D88F-3365-6E2B-D86615BF7C37 is not found. But that uuid is the Device uuid and 55001000-ABCD-429F-9150-4BF2A5ED8308 is the Service uuid.

Steps to reproduce the behavior:
1. Connect to one device connectToDevice with device ID: 4C57CE91-D88F-3365-6E2B-D86615BF7C37
2. Read from characteristic  uuid: 00002430-CAFE-4BED-A517-6AEEBA9EEACA with service uuid: 55001000-ABCD-429F-9150-4BF2A5ED8308 and device uuid: 4C57CE91-D88F-3365-6E2B-D86615BF7C37
3. 

Expected behavior:
Return characteristic data

- iPhone iOS 14.1
- Flutter 1.22.3

werediver, 2020-11-05:
If the error message says a _service_ with the specified ID is not present / has not been discovered, whereas you expect that ID to specify a _device_, the most likely reason for that is you put the device ID in place of a service ID. Please, double-check and provide the relevant code, if you don't find a mistake.

HelloBytes, 2020-11-05:
Thank you for your fast answer!

Double checked

QualifiedCharacteristic _c = QualifiedCharacteristic(
          deviceId: "4C57CE91-D88F-3365-6E2B-D86615BF7C37",
          serviceId: Uuid.parse("55001000-ABCD-429F-9150-4BF2A5ED8308"),
          characteristicId: Uuid.parse("00002430-ABCD-4BED-A517-6AEEBA9EEACA"));

 List<int> response = await _ble.readCharacteristic(_c);

Could it be that the device does not proper return the discovered services / characteristics?

werediver, 2020-11-05:
It seems that the service ID and the peripheral ID are swapped (misplaced) in the exception description (on iOS). This will be fixed.

How do you connect to the device? Are you sure you read the characteristic only after the connection is established as per `connectToDevice(⋯)`? (seeing the relevant code could help)

HelloBytes, 2020-11-05:
I indeed read the characteristics after connecting. servicesWithCharacteristicsToDiscover makes no difference:

`_connection = _ble
        .connectToDevice(
      id: uuid,
//      servicesWithCharacteristicsToDiscover: {
//        _fS.uuid: [_sC.uuid],
//      },
      connectionTimeout: const Duration(seconds: 2),
    )
        .listen((e) {
      print("BLE DFGT: connecting $e");
    }, onError: (Object error) {
      print("BLE DFGT: connecting error $error");
    }, onDone: () {
      print("BLE DFGT: connecting done");
    });`

I receive the DeviceConnectionState.connected from connectedDeviceStream, after this I read the characteristic. In android there is no problem.

werediver, 2020-11-05:
You are passing an empty map to `servicesWithCharacteristicsToDiscover:`, which means "discover no services". You should either pass `null` (or omit this parameter entirely), of pass the actual services and characteristics you're going to work with (the preferred way).

HelloBytes, 2020-11-05:
I found something, If I do manualy:
var test = await _ble.discoverServices(uuid);

Its returns: 

DiscoveredService(serviceId: 55001000-cafe-429f-9150-4b2fa5ed8308

But it has to be 55001000-cafe-429f-9150-4bf2a5ed8308

55001000-cafe-429f-9150-4b2**fa**5ed8308
55001000-cafe-429f-9150-4bf**2a**5ed8308

werediver, 2020-11-05:
Please pay attention to my previous comment.

Regarding the ID mismatch, you should trust the data reported (by the system / BLE stack) through the library. We don't mess with the IDs.

HelloBytes, 2020-11-05:
You where right, sorry for your time and thanks for the great plugin. Our documentation was incorrect. But strange it worked on android with the wrong service id.

closed 2020-11-05 by HelloBytes