PhilipsHue/flutter_reactive_ble #140: It is possible to use  flutter_reactive_ble  to send and receive data between an Android device and an iOS device through BLE?  if yes how

develop11xt, 2020-11-11:


remonh87, 2020-11-13:
In order to communicate through BLE you need to have 2 devices:

**Central**=> simple said the client device that reads and writes data from another BLE device.
**Peripheral**=> the device that is advertising and acts as the server.

In most use cases phones are acting as Centrals and we use this to connect to other devices. Our library is made with this use case in mind. However if you are able to change a phone into a BLE peripheral you can use this library to connect to it and interact with some of the GATT services that you defined. 

How to do this see:
- Android => https://devzone.nordicsemi.com/f/nordic-q-a/26730/can-an-android-device-be-a-peripheral
- iOS => https://blog.usejournal.com/creating-ios-application-as-bluetooth-peripheral-669404230232

However there are more easy ways to send data from one phone to the other (there are nowadays lots of cloud services), so I would double check if BLE is the right way to communicate between devices.

werediver, 2020-11-16:
See also #49.

closed 2020-11-16 by werediver