PhilipsHue/flutter_reactive_ble #142: subscribeToCharacteristic crash IOS application

kapiios, 2020-11-12:
I tested 3 scenarios:
1. just **subscribeToCharacteristic** and tried to listen notyfication on characteristic but imbecilely application crash

2. first step was add listen on the characteristicValueStream but only print the values to have a listener:
 `     _flutterReactiveBle.characteristicValueStream.listen((event) {
      // print(event.result);
    });`

then subscribe to chratetistic and listen to:
`Stream<List<int>> stream = _flutterReactiveBle.subscribeToCharacteristic(characteristic)`

for this it works fine and stream and the characteristicValueStream give the same data

3. I modify the package by myself and add to turn on notification on characteristic 
`  Future notifyCharacteristic(QualifiedCharacteristic characteristic) async {
    await _pluginController.subscribeToNotifications(characteristic).first;
  }`

then in my application I first listen on characteristicValueStream
`    Stream<List<int>> stream = _flutterReactiveBle.characteristicValueStream
        .asyncMap((data) => data.result.dematerialize());`
and turn on notification 
`_flutterReactiveBle.notifyCharacteristic(characteristic);`

the workaround in this way for me is more readable because it is informing that it turn on notification, subscribeToCharacteristic not do that and developer need to deep into code to find it out that it turn notifications.

To me the problem is connected with that the notification are turn on before start listening to characteristic and it crash the app but I'm not bluetooth expert.

werediver, 2020-11-16:
@kapiios Please, report the exact version of the library you're experiencing the issue with.

Seems to duplicate #137 (which we're not sure is fixed yet). @dawidb-appdate did you have a chance to do the extra checks you [mentioned](https://github.com/PhilipsHue/flutter_reactive_ble/issues/137#issuecomment-726304067)?

kapiios, 2020-11-17:
Hello @werediver i was testing it with version 2.6.1+1, but after update to 2.7.1 it still crashing with assertion:
Assertion failed: file flutter_reactive_ble/PluginController.swift, line 103

if it dose matter I'm using iPhone with IOS 14

werediver, 2020-11-18:
@kapiios Thank you for checking on a newer version. Let's track this issue in #137, which was opened earlier.

closed 2020-11-18 by werediver