PhilipsHue/flutter_reactive_ble #147: BLE unable to discover and connect device immediately after iOS app (in release mode) launches.

0100101101001011, 2020-11-19:
**Describe the bug**
I have stored a BLE device UUID in the app local database and try to connect to it right after the flutter app launches. Everything works alright on Android (Both Debug and Release Mode) and iOS debug mode ( when manually start the app in the Xcode).

However, when I test the app in the iOS release mode, the BLE doesn't seem to be able to make the connection immediately after the app starts. I have to call it seconds later to make the connection.

werediver, 2020-11-20:
@0100101101001011 Are you doing `WidgetsFlutterBinding.ensureInitialized()` first thing in your `main()`?

```dart
Future<void> main() async {
  WidgetsFlutterBinding.ensureInitialized();
  
  // ...
}
```

If not, does that fix the issue for you?

werediver, 2020-11-20:
Actually, it seems you just have to wait till the powered-on status is _actually reported_ by the API. Until then it's not allowed to perform BLE operations.

0100101101001011, 2020-11-21:
Thank you for your help! It solved the problem:) Why doesn't it occur on Android though?

werediver, 2020-11-21:
I'm glad your issue is solved. I hope the same approach works for you on both iOS and Android and you need no platform checks?

The BLE stack implementations on iOS and Android are entirely different, even though they implement the same protocols. They may (and do) have some discrepancies and _different_ limitations; this turns out to be one of them.

Let us keep the issue open and document the necessity to await for powered-on status before making BLE calls in README. It should be useful to many users of the library.

CC @remonh87

closed 2021-07-04 by remonh87

v11, 2022-11-28:
I had the same issue in iOS in release mode. It is not super well stated in the documentation.

Just in case somebody needs the code:

```
void waitBluetoothReady() async {
  statusStream = flutterReactiveBle.statusStream.listen((event) {
    switch (event) {
      // Connected
      case BleStatus.ready:
        {
          // Ready
          // Start scanning
          // ...
          break;
        }
      default:
    }
  });
}
```

leemind, 2023-02-08:
> I had the same issue in iOS in release mode. It is not super well stated in the documentation.
> 
> Just in case somebody needs the code:
> 
You have just saved the day! Huge thanks

PR #305 (2021-07-04): add ensuring ble stack ready to readme
Fix #147