PhilipsHue/flutter_reactive_ble #153: Long term support and Future?

realrk69, 2020-11-30:
I've been looking at various libraries for connecting IoT devices developed by us to smartphones. We previously used native android which has been very stable and errors are predictable, but now have a need to develop for iOS and expand to windows and Linux too. I have a few questions here:

1. Will this library have long-term maintenance or support? 
2. Does Phillips Hue actually use this library?
3. Is there any plan to release support for Web BLE? That way all platforms (except iOS) will automatically be enabled for our user base.
4. Are there any serious platform/performance considerations or app architecture recommendations with this app? (please link threads if possible)

I've been integrating Flutter_Blue (thought of Polidea's too) with my app but can switch to this if stability is good here. I don't want to build something and have it outdated within a year. Let me know your thoughts. Thanks!

PS - I think all open source libraries should have the above-mentioned questions' answers.

remonh87, 2020-11-30:
@realrk69  See below the answer to your questions

1. We use the library ourselves for our Philips Hue Bluetooth app and Philips Hue sync App. Since these apps are used by our customers we are actively maintaining it. Also it is an open source project and we encourage the Flutter community to help us out making our library better.

2. Yes we use it for 2 apps.

3. At this point we do not have plans to develop support for Web BLE ourselves however if others are willing to add functionality for web BLE to this library we are willing to support them.

4.In general I can say that BLE on Android is more tricky due to fragmentation than on iOS but our library implemented some tricks to handle those. In regard to architecture: feel free to use what you like. Within Philips Hue we use [Redux](https://pub.dev/packages/redux) (for managing state) in combination with [Provider](https://pub.dev/packages/provider) (propagating data to our widgets).  Our library is very well documented so the usage should be quite clear, if not please file an issue since we are very keen on having clear documentation.

realrk69, 2020-12-01:
Thanks for the quick response. I'm using Provider (ChangeNotifier) only for both managing state and data propagation. Can I get some documentation or links to know a little bit more about handling abrupt disconnections from the device or the user himself and how it can be handled (if inbuilt)?

remonh87, 2020-12-01:
When you connect to a device on our library we return a stream that propagate connectionstateupdates for that device. By incorporating that for example with a StreamProvider you can easily update your UI based on a new value.

See also our example app which has a very basic solution: https://github.com/PhilipsHue/flutter_reactive_ble/tree/master/example

closed 2020-12-02 by remonh87