PhilipsHue/flutter_reactive_ble #157: Disconnect an already connected device

ollydixon, 2020-12-02:
There doesn't seem to be anyway to disconnect a device that was previously connected via bluetooth. (without knowing the device id)

remonh87, 2020-12-02:
Disconnecting a device is be done by terminating the streamsubscription of the connected device. Also make sure check our example app and the documentation of the `connectToDevice`  method for more advanced use cases.

closed 2020-12-02 by remonh87

safield, 2021-01-29:
> Disconnecting a device is be done by terminating the streamsubscription of the connected device. Also make sure check our example app and the documentation of the `connectToDevice` method for more advanced use cases.

How do you know when the disconnect process has completed, if the action of inititating disconnect actually involves unsubscribing from the very stream intended to communicate when transitions to disconnecting and disconnected states occur. 

This seems like an inherent design flaw.

werediver, 2021-01-30:
Why would you need to _immediately_ connect to device you have _intentionally_ disconnected from?

In 99.99% of use cases you can assume the device is disconnected immediately after you canceled the subscription to the connectionn-state stream (because you're not going to _immediately_ reconnect to it, after you have _intentionally_ disconnected from it).

In the 0.01%  of use cases, where you actually need to know when the disconnecting is completed, you can listen the multiplexed connection-status stream [FlutterReactiveBle.connectedDeviceStream](https://github.com/PhilipsHue/flutter_reactive_ble/blob/dcbf292b25ccaec542c3510197820eaa354b8b80/lib/src/reactive_ble.dart#L70).

safield, 2021-02-01:
I am writing an wrapper around this bluetooth library. I prefer to fully capture all states, and want to make an explicit guarantee to users of my class when and when not it is safe to connect (thats just called good practice). I don't assume what their use cases are, but most importantly, this lack of rigour in design will lead to more obscure edge cases (bugs) that I generally prefer to prevent.

Thanks for pointing me to connectedDeviceStream!

> Why would you need to _immediately_ connect to device you have _intentionally_ disconnected from?
>
> In 99.99% of use cases you can assume the device is disconnected immediately after you canceled the [...]

I am writing an wrapper around this bluetooth library. I prefer to fully capture all states, and want to make an explicit **guarantee** to users of my class when and when not it is safe to connect (thats just called good practice). I don't assume what their use cases are, but most importantly, this lack of rigour in design will lead to more obscure edge cases (bugs) that I generally prefer to prevent.

Thanks for pointing me to connectedDeviceStream I will use it (and so should you!) :)