PhilipsHue/flutter_reactive_ble #159: Migrating from other BLE libs

realrk69, 2020-12-02:
Hi there, due to stability issues with the previous library I was working with (flutter_blue), I'm trying to migrate the library to reactive ble. Need some help. Could someone indicate what I'm doing wrong here?

```
import 'package:flutter/cupertino.dart';
import 'package:flutter/material.dart';
import 'package:flutter_reactive_ble/flutter_reactive_ble.dart';
import 'package:flutter_spinkit/flutter_spinkit.dart';
import 'package:flutter_staggered_grid_view/flutter_staggered_grid_view.dart';
import 'package:provider/provider.dart';
import 'package:app/providers/ble_status_monitor.dart';
import 'package:app/providers/reactive_state.dart';
import 'package:app/providers/ble_scanner.dart';
import 'package:app/providers/ble_device_connector.dart';

class BleConnectionScreen extends StatelessWidget {
  @override
  Widget build(BuildContext context) {
    
    final BleStatus status = Provider.of<BleStatus>(context);
    final BleScannerState scannerState = Provider.of<BleScannerState>(context);
    final bleScanner = Provider.of<BleScanner>(context);
    final connectionUpdate = Provider.of<ConnectionStateUpdate>(context);
    final deviceConnector = Provider.of<BleDeviceConnector>(context);

    //bool _deviceConnected() =>
    //    connectionUpdate.connectionState == DeviceConnectionState.connected;

    ListView _buildListViewOfDevices(devicesList) {
      List<Container> containers = new List<Container>();
      for (DiscoveredDevice device in devicesList) {
        containers.add(
          Container(
            height: 50,
            child: Row(
              children: <Widget>[
                Expanded(
                  child: Column(
                    children: <Widget>[
                      Text(
                          device.name == '' ? '(unknown device)' : device.name),
                      Text(device.id.toString()),
                    ],
                  ),
                ),
                FlatButton(
                    color: Colors.blue,
                    child: Text(
                      'Connect',
                      style: TextStyle(color: Colors.white),
                    ),
                    onPressed: () {
                      deviceConnector.connect(device.id);
                      // Show Connected in Status
                    }),
              ],
            ),
          ),
        );
      }

      return ListView(
        padding: const EdgeInsets.all(8),
        children: <Widget>[
          ...containers,
        ],
      );
    }

    String _bleConnectionAccess(BleStatus status){
        switch (status) {
[... 159 lines cut ...]
```
════════ Exception caught by rendering library ═════════════════════════════════════════════════════
The following NoSuchMethodError was thrown during performLayout():
The getter 'key' was called on null.
Receiver: null
Tried calling: key

The relevant error-causing widget was: 
  StaggeredGridView file:///C:/Users/realr/Desktop/RAAV/app/lib/screens/BleConnectionScreen.dart:136:34
When the exception was thrown, this was the stack: 
#0      Object.noSuchMethod (dart:core-patch/object_patch.dart:51:5)
#1      SliverChildListDelegate.build (package:flutter/src/widgets/sliver.dart:679:27)
#2      SliverVariableSizeBoxAdaptorElement._build.<anonymous closure> (package:flutter_staggered_grid_view/src/widgets/sliver.dart:144:38)
#3      _HashMap.putIfAbsent (dart:collection-patch/collection_patch.dart:140:29)
#4      SliverVariableSizeBoxAdaptorElement._build (package:flutter_staggered_grid_view/src/widgets/sliver.dart:143:26)
...
The following RenderObject was being processed when the exception was fired: RenderSliverStaggeredGrid#74d89 relayoutBoundary=up2 NEEDS-LAYOUT NEEDS-PAINT NEEDS-COMPOSITING-BITS-UPDATE
...  parentData: paintOffset=Offset(0.0, 0.0) (can use size)
...  constraints: SliverConstraints(AxisDirection.down, GrowthDirection.forward, ScrollDirection.idle, scrollOffset: 0.0, remainingPaintExtent: 698.9, crossAxisExtent: 360.7, crossAxisDirection: AxisDirection.right, viewportMainAxisExtent: 714.9, remainingCacheExtent: 948.9, cacheOrigin: 0.0)
...  geometry: null
...  currently live children: 0,1,2
RenderObject: RenderSliverStaggeredGrid#74d89 relayoutBoundary=up2 NEEDS-LAYOUT NEEDS-PAINT NEEDS-COMPOSITING-BITS-UPDATE
```

Also, is this an issue with my provider implementation? or with the UI rendering?

remonh87, 2020-12-02:
Based on the stracktrace this looks like an issue with the staggeredgridview.. The providers that you declare look good I do not see any issues with those despite I personally prefer to use selector (in combination with a viewmodel) to have more granular widget rebuilds. I think it makes sense to file this issue to the StaggeredGridView library.

realrk69, 2020-12-02:
You're right, it was staggeredview since I mistakenly upgraded to their null safety version which is in alpha. BTW great work on this library, let me know if I can help with Web BLE implementation. I've worked a lot on JS previously and can help in writing the wrappers for V8.

closed 2020-12-02 by realrk69