PhilipsHue/flutter_reactive_ble #160: WriteCharacteristicWithResponse returns void

realrk69, 2020-12-02:
I think this is a bug:

```
final characteristic = QualifiedCharacteristic(serviceId: serviceUuid, characteristicId: characteristicUuid, deviceId: foundDeviceId);
final response = await _ble.writeCharacteristicWithResponse(characteristic, value: [0x00]);
```
It says the response is void and hence cannot be printed. What's wrong here?
In the documentation, it says the response is a List<int>

remonh87, 2020-12-03:
The `List<int>` is a mistake in the readme which I will fix. Write with acknowledge just awaits a response for the device and that is it. The response is just an acknowledge call. BLE does not have a request response mechanism implemented like for example HTTP. See also #150  for a more elaborative answer.

closed 2020-12-03 by remonh87

werediver, 2020-12-03:
See also #160, exactly the same confusion.

realrk69, 2020-12-03:
What about the very long connect duration? It doesn't happen always...

werediver, 2020-12-03:
That is unfortunate you're mixing two separate issues in one ticket.

You're using an Android device and they are known to exhibit very different BLE stack performance depending on their vendor, model, Android version, and your application behavior (the latter can partially [mostly?] be patched by the library you're using, Flutter Reactive BLE and Polidea RxAndroidBle in this case).

You may want to explicitly specify a reasonably short connection timeout (think of tens of seconds, I wouldn't recommend going below 20–30 seconds on Android) when calling `connect(⋯)` (please, check the documentation for the method). Then, in case of failure, you can retry.

realrk69, 2020-12-03:
I think it's an application side error. Since my previous library was also having this problem (flutter blue) but I fixed it by listening to ScanResults while connecting to the device. Will have to see how to implement it here.

werediver, 2020-12-03:
You should _not_ run discovery when connecting to peripherals (i.e. only _before_ that) from Android devices. Android BLE stack often cannot handle this well.

realrk69, 2020-12-03:
I tested on 2 devices and it seems to connect to esp32 easily compared to nrf. Is there any way to change the scan mode to low latency or balanced?

werediver, 2020-12-03:
You should check the API its documentation in the first place: [reactive_ble.dart#L233](https://github.com/PhilipsHue/flutter_reactive_ble/blob/f55da67d0c9ed977741730979d4122978a953066/lib/src/reactive_ble.dart#L233)

realrk69, 2020-12-03:
Got it, I was not disconnecting the device before relaunching the app. Actually, the complication is that I'm using a single page for managing everything related to connections, so it takes some optimization. Thanks for your help