PhilipsHue/flutter_reactive_ble #164: Feature question

wolfc01, 2020-12-03:
Hi,

in a pyboard rev d i'm running a script which implements a BLE UART service (UUID: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E), see https://github.com/micropython/micropython/blob/master/examples/bluetooth/ble_simple_peripheral.py for details

My Question: is it in principle possible using the flutter_reactive_ble lib to create an UART counterpart on an android smartphone? Now i'm using the well known android test app of Nordic: nRF UART V2.0.

Best regards
Carl.

remonh87, 2020-12-06:
@wolfc01 if I understand your question correctly you want to use the Android phone to connect to the peripheral that you created. This is possible with our library. You can already test it out by running the example app and see if you are able to connect to the device.

closed 2020-12-06 by remonh87

wolfc01, 2020-12-07:
@remonh87 

Thanks, I will try out the example.

Br
Carl.