PhilipsHue/flutter_reactive_ble #166: Waiting for stream to end before returning List from Stream

realrk69, 2020-12-05:
So I have a StreamSubscription that correctly reads data received from my device.

```
StreamSubscription<List<int>> _rxSubscription;
List<List<int>> _response = new List<List<int>>();

_rxSubscription = _ble.subscribeToCharacteristic(_rx).listen((event) {
      _response.add(event);
    });

writeToDevice(List<int> command) async {
    _rxSubscription.resume();
    await _ble.writeCharacteristicWithoutResponse(_tx, value: command);
    return _response;
  }
```
Here is how the device responds:

```
I/flutter ( 5506): [0, 65, 0, 0]
I/flutter ( 5506): [0, 0, 104, 220, 69, 91, 240, 63, 0, 0, 240, 100, 112, 209, 240, 63, 0, 0, 176, 19]
I/flutter ( 5506): [133, 233, 240, 63, 0, 0, 104, 126, 172, 204, 240, 63, 0, 0, 96, 250, 189, 193, 240, 63]
```
This data is continuously streamed when any write operation is performed and can vary in length depending on the 'command' meaning there can be n number of packets which are sent back. Right now after running the function, it returns an empty list of lists since _response is returned before the device starts responding.

**What I'm trying to do:** Get the _response to return only after all the data is collected and the device stops responding, ie. the Stream stops streaming. How would I do this in dart? I don't want to use a timer, since the data might take longer or shorter.
(Currently there is no feature of the packets which might indicate 'end of line' of 'end of message')

realrk69, 2020-12-06:
Any help is appreciated!

remonh87, 2020-12-08:
@realrk69  first question that I have is how do you know you have received the last packets (does your peripheral emit closing bytes)? I think a completer https://api.dart.dev/stable/2.10.2/dart-async/Completer-class.html can help for your use case.  If you are familiar with reactive programming you can also use https://pub.dev/packages/rxdart .

realrk69, 2020-12-08:
The device doesn't emit closing bytes (according to the datasheet), but it does declare how many packets need to be captured in the first packet itself.

For example in the above output from the device, the first packet:
[0, 65, 0, 0]
First 0 => success, 1 would indicate failure
65 => total number of packets that need to be captured after this packet, with each packet having 20 bytes.

realrk69, 2020-12-08:
[image]

This image is from the vendor

realrk69, 2020-12-12:
Let me know if this doc was helpful. Am still trying to fix this. Currently, am using a Timer with async to wait for the packets for a predetermined time. But if this is somehow possible without using an unreliable timer, I'm game (since some packets may be longer or shorter depending on the command sent)

remonh87, 2020-12-12:
@realrk69 the issue you reported has nothing to do with our library I gave you some suggestions on how to solve the problem using some standard dart async programming solutions.

For example you can do something like this (I use some pseudo code) :

```dart
final completer = Completer<void>();
rxSubscription = _ble.subscribeToCharacteristic(_rx).listen((event) {
      _response.add(event);
         if(lastPacket){
            completer.complete();
        }
    });
// write logic

return completer.future();
```

closed 2020-12-12 by remonh87