PhilipsHue/flutter_reactive_ble #167: scanForDevices only shows device once

ollydixon, 2020-12-06:
scanForDevices only shows the device once, we need to reset the device for it to show again.

The following code when opening the screen

```
_flutterBlue = FlutterReactiveBle();
_bluetoothSubscription = _flutterBlue.scanForDevices(
    withServices: [Uuid.parse(MUSE_SERVICE)],
    scanMode: ScanMode.lowLatency).listen((device) {
      // Devices, but they don't show again on opening this screen again.
});
```

When we close the widget we also 

```
    _flutterBlue?.deinitialize();
    _bluetoothSubscription?.cancel();
```

No luck, any ideas?

remonh87, 2020-12-08:
@ollydixon do you display the devices that are emitted by the stream on the screen? In that case that would explain the behaviour. For a proper scan implementation checkout https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/example/lib/src/ble/ble_scanner.dart .  Also can you try if you see the device with our example app?

ollydixon, 2020-12-08:
@remonh87 thanks for the relpy. I used the examples and I reset the instance of Flutter BLE even with deinitlize. Going back to the same screen; it still doesn't show the device. The device is reset, powered on off, nothing. Very strange. I think it's an issue with the stream now updating when we reinitialize the BLE instance, also it seems when we end BLE it doesn't actually disconnect the device at all times.

remonh87, 2020-12-10:
I think the issue that you are facing with deinitialize has to do with the fact you are not awaiting the deinitialization.  Try 
`await  _flutterBlue?.deinitialize();` . Regarding the instances not visible: can you share some of the code of the screen ? Are you using a streambuilder or provider to provide the data?

ollydixon, 2020-12-10:
@remonh87 I tried that as well, we also have ```_flutterBlue?.deinitialize()```; on the dispose methods. No luck.

ollydixon, 2020-12-10:
@remonh87 just doing it inside a simple init, then pushing the data to a Map (print on the map calls) very simple setup. I can't share the code unfortuntely there's some sensative stuff in there.

But, looking at the sample app code we get the same issue, we have to restart the app to get the connection back on iOS.

remonh87, 2020-12-10:
@ollydixon just a small question do you experience the same behaviour on an Android device? Just to narrow things down

ollydixon, 2020-12-10:
@remonh87 I'll get back to you on that. 👍

ollydixon, 2020-12-10:
@remonh87 this is going to sound crazy. But after a Flutter clean it works fine...

ollydixon, 2020-12-10:
@remonh87 thanks for the help though.

closed 2020-12-10 by ollydixon

remonh87, 2020-12-10:
That is very interesting feel free to re-open the issue once it happens again. Happy you could continue though :)