PhilipsHue/flutter_reactive_ble #168: Building with Gradle 6.5 fails

realrk69, 2020-12-06:
**Describe the bug**
Gradle build with version 6.5 (distributionUrl=https\://services.gradle.org/distributions/gradle-6.5-all.zip) in gradle wrapper properties.

Stack trace:
```
Running Gradle task 'assembleDebug'...

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':flutter_reactive_ble'.
> Failed to notify project evaluation listener.
   > com.android.build.gradle.internal.variant.BaseVariantData.getScope()Lcom/android/build/gradle/internal/scope/VariantScope;

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 7s
Exception: Gradle task assembleDebug failed with exit code 1
```

The build fails continuously

remonh87, 2020-12-08:
I can compile the example app of our library with Gradle 6.5 without problems. However it can be other dependencies that you are using are causing this issue. Above issue is normally related to the fact you are using a outdated Android tools in combination with a new gradle version or the other way around.

Also keep in mind that Flutter themselves are in still not on Gradle 6 and Android tools 4 [issue link](https://github.com/flutter/flutter/issues/49438) . I would be very cautious with upgrading to latest greatest with Flutter until the mentioned issue is in stable.

closed 2020-12-08 by remonh87

realrk69, 2020-12-08:
Understood. Thanks for the detail!