PhilipsHue/flutter_reactive_ble #170: Example code

wolfc01, 2020-12-09:
Hi,

I've seen the example code, but i'm still a bit puzzled to get started with this ble library. Is it possible to share some more examples or code snippets which e.g. sends and retrieve some messages, in order to get me on track?

Br 
Carl

wolfc01, 2020-12-09:
Hi

I found already a piece of example code at 

https://1littleendian.medium.com/the-late-night-tinkering-project-11-arduino-flutter-e02586fd62ac

Br
Carl.

remonh87, 2020-12-10:
@wolfc01  thank you for your message. We have some examples in our readme file of the repo, also our public API has a lot of documentation. What in particular did you miss and made you struggle?  Let us know we are always looking to improve :)

wolfc01, 2020-12-10:
Hi Remon,
thanks for the quick reply!

The documentation is excellent, no problem with that, for a seasoned Flutter/BLE coder. 

As a relative newbie in Flutter/BLE and being a decision maker, I have to decide if this library is the way to go for our project.  

A more "full example" covering the following items would help:
- Connect/disconnect (is already there in example)
- Connection state monitoring (is already there in example)
- Search for device / services (is already there in example)
- Sending a message to characteristic of xx bytes (bytes entered in a entry box)
- Receiving a message from characteristic (bytes received shown in text box)
- The latter two using response and no response.

I hope the above expresses my "requirements" for extending the example.

Best regards,
Carl.

farr64, 2020-12-10:
I concur with Carl's opinion. Being able to detect and connect is great BUT being able to send messages back and forth is THE reason for connecting (after having detected available devices).

So, for mere mortals, simple examples of actual communication (with "meaning", as Claude Shannon mentioned -- in addition to just "the channel") are extremely useful.

Thanks very much, @wolfc01, for the reference:
https://1littleendian.medium.com/the-late-night-tinkering-project-11-arduino-flutter-e02586fd62ac

Thanks again, @remonh87 and @werediver,  for such a wonderful library and for such responsive support.

wolfc01, 2020-12-10:
All,

Additionally, a bit more elaborated example will help greatly to grow popularity of this library. 

I studied other flutter ble libraries which i think are less mature than flutter_reactibe_ble. 

Currently Flutter suffers from a lack of a standard  wide accepted and good maintained ble library. 

Given the use case of flutter_reactive_ble (hue lighting) and the  continuous maintenance done on flutter_reactive_ble I think this library has a chance to become the defacto standard. Again: good examples surely will help this library further lift off!

Br
Carl

dianemarquette, 2020-12-14:
Hi,

Being relatively new to Flutter and BLE, I'm also struggling to find an example with **requestMtu**. I saw the code snippet in the ReadMe. However, it doesn't seem to be included in the demo app of the library.

Do you know where I should integrate it to the code? I've been stuck for days and I keep getting this error "error: The argument type 'Future<int> Function({String deviceId, int mtu})' can't be assigned to the parameter type 'Future<int> Function(String, int)'. (argument_type_not_assignable at [checky] lib/main.dart:225)".

Thanks for your time.
Best,

Diane

farr64, 2020-12-27:
Your README describes how to do seven important things:

1. Initialization
2. Device discovery
3. Observe host device BLE status
4. Establishing connection
5. Read / write characteristics
6. Subscribe to characteristic
7. Negotiate MTU size

Your example provides very nice guidance for the first four, but does not provide examples for the last three:

5. Read / write characteristics
6. Subscribe to characteristic
7. Negotiate MTU size

If you could please consider "upgrading" your example to provide some simple sample code for reading and writing (including subscribing), I'm sure everyone would be very grateful. If you were to include an example for negotiating MTU size, I'm sure Diane would be most grateful.

As a referemce, this provides some limited examples for writing. I gave up trying to make it work for reading:
https://github.com/flutter-devs/flutter_bluetooth

Thank you very much,

Alfredo

wolfc01, 2020-12-27:
Hi,

I created a pull request to help building example code, please refer to https://github.com/PhilipsHue/flutter_reactive_ble/pull/178

Best regards
Carl.

wolfc01, 2020-12-28:
Hello,

@farr64 @dianemarquette: the reference to my ble example has been added  to the README.md of the flutter_reactive_ble library.

Review comments welcome.

@remonh87 : should we close this issue, or is there in your opinion more work to do? 
In my opinion my example needs to be thoroughly reviewed by one of your guys, as copy-paste this example as a base for other apps will also copy-paste structural errors if such errors exist in the example. Such errors do not help with growing of the flutter_reactive_ble library.

Br
Carl.

remonh87, 2021-01-04:
I think for now it can be close thanks for your contribution.

closed 2021-01-04 by remonh87