PhilipsHue/flutter_reactive_ble #171: Device erroneously reported as disconnected after app restart

ElHyperion, 2020-12-10:
**Describe the bug**
When restarting the app after having connected to a device and leaving the connection opened, the `connectedDeviceStream` reports immediately after the restart that the device has been disconnected, while apparently it is still connected. I found out it is still connected by scanning for new devices after the restart, where the device won't show up anymore, and turning the connected device off again soon results in `connectedDeviceStream` events that the device is disconnecting, and later that it has been disconnected. Rerunning the scan afterwards successfully finds it (if the device has been powered on again of course).

I cannot use the library to disconnect from the device after the restart because I have no stream subscription to it from the `connectToDevice()` stream and the only solution is to either restart the device, or restart my phone's Bluetooth.

Tested on debug and release mode, same behaviour observed on both.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one device using `connectToDevice()`
2. Restart the application
3. Listen to the `connectedDeviceStream()`
4. Observe that the ConnectionStateUpdate(deviceId: XX:XX:XX:XX:XX:XX, connectionState: DeviceConnectionState.disconnected, error: null) event has been received
5. Shut down the device to observe that a DeviceConnectionState.disconnecting event together with .disconnected afterwards is received soonly, and that the device can be scanned again afterwards it is started again, but not before it has been shut down before.

**Expected behavior**
After restarting the app, the `connectedDeviceStream()` should report a DeviceConnectionState.connected value instead.

**Peripheral device**
- Vendor, model: ESP32 
- Firmware: ESP-IDF V3

werediver, 2020-12-10:
You do not specify on which mobile platform you are experiencing the issue, please clarify.

What do you mean by app restarting? Hot reload, hot restart, or "cold" restart (full termination and start)?

At least if we are talking about a cold restart the (logical) connection to your peripheral is considered terminated, so the mobile app should establish a new connection to the peripheral, as usual.

ElHyperion, 2020-12-10:
Sorry for not including the platform information, added it (I'm on Android). I'm restarting the app through Flutter: Hot Restart command. The other BT libraries I tested (flutter_blue and flutter_ble_lib) didn't have problem with a hot restart while being connected, so I automatically expected the same behaviour here. So the correct way should be to perform a cold restart every time?

Just tested with a cold restart (terminating the app and starting it again) and it seems to work fine, and it's not reconnecting automatically. Is this expected behaviour and should I implement auto reconnect on my own?

werediver, 2020-12-10:
When you connect to a peripheral with this library you get a stream reporting the connection status. When you cancel the subscription to this stream, this is a command to disconnect. So, no stream, no connection.

Both iOS and Android can _cache_ and _share_  connections to BLE peripheral, hence the fact that an app needs to call some kind of "connect" method does not always mean the system will need to (re-)establish a connection, it may already have one and will just pass it through to the application.

When the connection actually needs to be established it has a power consumption impact, it may be considered to have some privacy impact, etc., so the app should decide if and when to reconnect. The library does not handle this.

We had an issue with hot reload on iOS in #10. I believe Android was handling this fine back then.

@remonh87 Do you expect any issues with hot restart on Android now? Or do you recognize this issue?

remonh87, 2020-12-10:
@ElHyperion can you send me some example code where it happens? I did use the example app and hot reload works (tested on Samsung s7)

ElHyperion, 2020-12-11:
> @ElHyperion can you send me some example code where it happens? I did use the example app and hot reload works (tested on Samsung s7)

Sure, I created a sample project to test the connectivity, using similar BT logic from my project.
Head over to my [flutter-ble-test repository](https://github.com/ElHyperion/flutter-ble-test) and try it out yourself (look for BT logic at services/flutter_reactive_ble_service.dart).

Using the sample project I just connected to a device, hot restarted the app, and the device was correctly recognized as connected and put into the connected devices list, but upon disconnecting from it I received a DeviceConnectionState.disconnected event while the device was still connected (it disappears from the list upon rescanning).

It never gets picked up by the scan until either the device or the phone's BT is turned off, or the app is properly restarted.

remonh87, 2020-12-18:
@ElHyperion  I did some triage today and for me it works like expected:

Hot restart
1. Scan for devices
2. Stop the scan
3. Establish a connection
4. Hot restart the app 
5. I noticed that the device is still connected with "Name pending"
6. I disconnect the device
7. Scan for devices again
8. The device shows in the list and I can connect to it.

Hot reload
1. Scan for devices
2. Stop the scan
3. Establish a connection
4. Hot reload the app 
5. I noticed that the device is still connected with the correct name
6. I disconnect the device
7. Scan for devices again
8. The device shows in the list and I can connect to it.

It looks like maybe there is something peculiar with the peripheral you are connecting to. Did you try it with another device?

remonh87, 2021-01-06:
I didn't receive a response will close the issue for now. Feel free to reopen if the issue still occurs

closed 2021-01-06 by remonh87

Artur110, 2021-01-29:
> * I noticed that the device is still connected with "Name pending"
> * I disconnect the device

Can you please explain how to detect connected device name and then disconnect it when the app just hot-restarted and stream subscriptions do not set yet (but device is connected)?

ElHyperion, 2021-01-29:
> I didn't receive a response will close the issue for now. Feel free to reopen if the issue still occurs
I am currently in the process of determining whether it is due to the hardware or not. I will probably get back to confirming / denying this issue, but it might probably take months until I get to it due to different project priorities now, so stay subscribed.

> > * I noticed that the device is still connected with "Name pending"
> > * I disconnect the device
> 
> Can you please explain how to detect connected device name and then disconnect it when the app just hot-restarted and stream subscriptions do not set yet (but device is connected)?
I do so by issuing the connect command again to the UUID, not sure if it's a correct approach though. But it seems to work and everyone else but me seems to use it without errors.

LiangHuangBC, 2021-06-24:
Got same error. Reason is if a device is connected before scan, you can not find it by scan. That's why when hot-restart clear the device list, and device disappeared in scan result.