PhilipsHue/flutter_reactive_ble #172: Unable to connect to iOS app

radivojeostojic, 2020-12-16:
I am getting this almost immediately after BleStatus.ready:

`Assertion failed: file flutter_reactive_ble/ConnectTaskController.swift, line 16
Assertion failed: file flutter_reactive_ble/ConnectTaskController.swift, line 16
* thread #1, queue = 'com.apple.main-thread', stop reason = Assertion failed
    frame #0: 0x00000001adc64e50 libswiftCore.dylib`_swift_runtime_on_report
libswiftCore.dylib`_swift_runtime_on_report:
->  0x1adc64e50 <+0>: ret    
libswiftCore.dylib`_swift_reportToDebugger:
    0x1adc64e54 <+0>: b      0x1adc64e50               ; _swift_runtime_on_report
libswiftCore.dylib`_swift_shouldReportFatalErrorsToDebugger:
    0x1adc64e58 <+0>: adrp   x8, 334858
    0x1adc64e5c <+4>: ldrb   w0, [x8, #0x4b0]
Target 0: (Runner) stopped.
Lost connection to device.
`

On Android everything works fine

werediver, 2020-12-23:
You are mentioning `BleStatus.ready`. Does this mean you are waiting till the BLE API interface is ready _before_ making _any_ further calls to the BLE API?

It looks like you may be making the "connect" call before the BLE interface is ready, at least I think that could lead to a similar issue.

remonh87, 2021-01-04:
I assume the issue is resolved feel free to reopen if this is not the case.

closed 2021-01-04 by remonh87