PhilipsHue/flutter_reactive_ble #173: ios Build error: 'flutter_reactive_ble/flutter_reactive_ble-Swift.h' file not found

florentin89, 2020-12-17:
**Describe the bug**
I imported the library and when I try to run my project I get the next error:

```
/Volumes/Kituri/Kituri/Flutter_SDK/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_reactive_ble-2.7.1/ios/Classes/ReactiveBlePlugin.m:2:9: fatal error: 'flutter_reactive_ble/flutter_reactive_ble-Swift.h' file not found
    #import <flutter_reactive_ble/flutter_reactive_ble-Swift.h>
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    note: Using new build system
    note: Building targets in parallel
    note: Planning build
    note: Constructing build description

Could not build the precompiled application for the device.

Error launching application on iPhone 7 Plus.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one / many devices using `connectTo()` / `connectToAdvertisingDevice()`
2. Read from / write to / subscribe to a protected / non-protected characteristic
3. Observe a failure with exception (including the part of the stack trace, belonging to this package) …

**Expected behavior**
A clear and concise description of what you expected to happen.

To run the app without errors because of this library.

**Smartphone / tablet**
 - Device: iPhone 7 plus
 - OS: iOS 14.1
 - Package version: flutter_reactive_ble 2.7.1

**Peripheral device**
- Vendor, model: iPhone 7 Plus
- Does it run a custom firmware / software: No

**Additional context**

```
Failed to build iOS app
Error output from Xcode build:
↳
    ** BUILD FAILED **

Xcode's output:
↳
    /Volumes/Kituri/Kituri/Flutter_SDK/flutter/.pub-cache/hosted/pub.dartlang.org/permission_handler-5.0.1+1/ios/Classes/strategies/PhotoPermissionStrategy.m:39:13: warning: enumeration value 'PHAuthorizationStatusLimited' not handled in switch [-Wswitch]
        switch (authorizationStatus) {
                ^
    /Volumes/Kituri/Kituri/Flutter_SDK/flutter/.pub-cache/hosted/pub.dartlang.org/permission_handler-5.0.1+1/ios/Classes/strategies/PhotoPermissionStrategy.m:39:13: note: add missing switch cases
        switch (authorizationStatus) {
                ^
    1 warning generated.
    /Volumes/Kituri/Kituri/Flutter_SDK/flutter/.pub-cache/hosted/pub.dartlang.org/permission_handler-5.0.1+1/ios/Classes/strategies/PhonePermissionStrategy.m:41:35: warning: 'subscriberCellularProvider' is deprecated: first deprecated in iOS 12.0 [-Wdeprecated-declarations]
        CTCarrier *carrier = [netInfo subscriberCellularProvider];
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                      serviceSubscriberCellularProviders
    In module 'CoreTelephony' imported from /Volumes/Kituri/Kituri/Flutter_SDK/flutter/.pub-cache/hosted/pub.dartlang.org/permission_handler-5.0.1+1/ios/Classes/strategies/PhonePermissionStrategy.m:8:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS14.3.sdk/System/Library/Frameworks/CoreTelephony.framework/Headers/CTTelephonyNetworkInfo.h:112:50: note: property 'subscriberCellularProvider' is declared deprecated here
    @property(readonly, retain, nullable) CTCarrier *subscriberCellularProvider API_DEPRECATED_WITH_REPLACEMENT("serviceSubscriberCellularProviders", ios(4.0, 12.0)) API_UNAVAILABLE(macos);
                                                     ^
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS14.3.sdk/System/Library/Frameworks/CoreTelephony.framework/Headers/CTTelephonyNetworkInfo.h:112:50: note: 'subscriberCellularProvider' has been explicitly marked deprecated here
    1 warning generated.
    /Volumes/Kituri/Kituri/Flutter_SDK/flutter/.pub-cache/hosted/pub.dartlang.org/permission_handler-5.0.1+1/ios/Classes/PermissionManager.m:69:59: warning: 'openURL:' is deprecated: first deprecated in iOS 10.0 [-Wdeprecated-declarations]
            BOOL success = [[UIApplication sharedApplication] openURL:[NSURL URLWithString:UIApplicationOpenSettingsURLString]];
                                                              ^~~~~~~
                                                              openURL:options:completionHandler:
    In module 'UIKit' imported from /Users/florentin/Desktop/VT/ios/Pods/Target Support Files/permission_handler/permission_handler-prefix.pch:2:
[... 391 lines cut ...]
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS14.3.sdk/System/Library/Frameworks/UIKit.framework/Headers/UIApplication.h:504:50: note: 'UIApplicationLaunchOptionsLocalNotificationKey' has been explicitly marked deprecated here
    UIKIT_EXTERN UIApplicationLaunchOptionsKey const UIApplicationLaunchOptionsLocalNotificationKey     NS_SWIFT_NAME(localNotification) API_DEPRECATED("Use UserNotifications Framework's -[UNUserNotificationCenterDelegate didReceiveNotificationResponse:withCompletionHandler:]", ios(4.0, 10.0)) API_UNAVAILABLE(tvos); // userInfo contains a UILocalNotification
                                                     ^
    /Volumes/Kituri/Kituri/Flutter_SDK/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_local_notifications-3.0.1+7/ios/Classes/FlutterLocalNotificationsPlugin.m:750:1: warning: implementing deprecated method [-Wdeprecated-implementations]
    - (void)application:(UIApplication*)application
    ^
    In module 'Flutter' imported from /Volumes/Kituri/Kituri/Flutter_SDK/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_local_notifications-3.0.1+7/ios/Classes/FlutterLocalNotificationsPlugin.h:1:
    /Volumes/Kituri/Kituri/Flutter_SDK/flutter/bin/cache/artifacts/engine/ios/Flutter.xcframework/ios-armv7_arm64/Flutter.framework/Headers/FlutterPlugin.h:98:1: note: method 'application:didReceiveLocalNotification:' declared here
    - (void)application:(UIApplication*)application
    ^
    35 warnings generated.
    /Volumes/Kituri/Kituri/Flutter_SDK/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_reactive_ble-2.7.1/ios/Classes/ReactiveBlePlugin.m:2:9: fatal error: 'flutter_reactive_ble/flutter_reactive_ble-Swift.h' file not found
    #import <flutter_reactive_ble/flutter_reactive_ble-Swift.h>
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    note: Using new build system
    note: Building targets in parallel
    note: Planning build
    note: Constructing build description

Could not build the precompiled application for the device.

Error launching application on iPhone 7 Plus.

```

remonh87, 2020-12-18:
@florentin89  thank you for your elaborate description. Can you also provide us the output of `flutter doctor -v` ?

florentin89, 2020-12-18:
Sure @remonh87. I will post it below:

```
MacBook-Pro:VT florentin$ flutter doctor -v
[✓] Flutter (Channel dev, 1.26.0-1.0.pre, on Mac OS X 10.15.6 19G36e darwin-x64, locale en-GB)
    • Flutter version 1.26.0-1.0.pre at /Volumes/Kituri/Kituri/Flutter_SDK/flutter
    • Framework revision 63062a6443 (5 days ago), 2020-12-13 23:19:13 +0800
    • Engine revision 4797b06652
    • Dart version 2.12.0 (build 2.12.0-141.0.dev)

[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.3)
    • Android SDK at /Users/florentin/Library/Android/sdk
    • Platform android-30, build-tools 29.0.3
    • Java binary at: /Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b3-6222593)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 12.3)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 12.3, Build version 12C33
    • CocoaPods version 1.9.3

[✓] Android Studio (version 4.0)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin version 49.0.2
    • Dart plugin version 193.7547
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b3-6222593)

[✓] VS Code (version 1.50.1)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.16.0

[✓] Connected device (1 available)
    • iPhone 12 Pro Max (mobile) • F5A40CCA-AA12-4C68-8214-2A232459193A • ios • com.apple.CoreSimulator.SimRuntime.iOS-14-3 (simulator)

• No issues found!
MacBook-Pro:VT florentin$ 

```

florentin89, 2020-12-18:
@remonh87 I managed to create a project from scratch where I'm getting exactly the same error when I run the command: `flutter build ios --debug`.
I will attach the link for this demo project, hoping that will help you to identify the issue. Also is a photo in that demo folder (screenshot_error.png).
Link to Demo Project:
[https://github.com/florentin89/test_ble](https://github.com/florentin89/test_ble)

Thanks,
Flo

remonh87, 2020-12-18:
@florentin89 I checked the project structure and it looks like you do not have swift support (your app is using objective-c). How to solve this issue follow these steps: https://github.com/ko2ic/image_downloader/issues/9#issuecomment-471608482

florentin89, 2020-12-18:
Thank you @remonh87 , you are a genius ! That was the fix ! You can close the issue now. Thanks again.

closed 2020-12-18 by remonh87