PhilipsHue/flutter_reactive_ble #174: Get rid of mockito

remonh87, 2020-12-18:
The problem is that Mockito needs codegen to comply to null safety. I would like to see if we can replace our mocks with stubs (also need to extract interfaces). This help us to not need to generate for our tests and we can start migrating and testing our library for null safety in Dart.

werediver, 2020-12-20:
Replacing the current mockito-dynamic approach with hand-written stubs is super tedious and would work much better if the stubs are generated. Now, mockito is going to offer such a stub-generator. What's wrong with it?

remonh87, 2020-12-20:
@werediver I did consider that option, had a discussion in the Flutter Study group about the best approach. The reasons for not choosing Mockito code generation  are:

-  It has still issues for example [bug](https://github.com/dart-lang/mockito/issues/297). Another dev mentioned it also does not work very well with the `@required` annotation.
- Mockito stil has no version that supports null-safety. We cannot migrate until they supply a version that supports it or at least update built_value dependency. 
- You can debate if codegen is the best option. Yes no manual labor but code generation in dart is slow and tedious, so maybe the dev experience will degrade for this.

I think for now I would propose for the manually created stub (that as you can see are quite simple). However we can consider later to code for codegen , however I did not see proper alternatives at the moment that support fully null-safety.

werediver, 2020-12-21:
It seems that your main argument is mockito provides a subpar null-safety support _yet_ and there is not good alternative _at the moment_ *and* you want to make the leap for null-safety _right now_, therefore we should switch to hand-written stubs.

Well, you did the job and you are willing to continue putting effort there, so on one hand go ahead. On the other hand, though, this hand-written mocks are really a low-end solution, I can tell from my experience. The biggest downsides I can highlight are:

- a lot of boilerplate code to type
- inconsistency between the hand-crafted stubs over time (you will find better ways to stub different things with time and forget some of them later, another dev will go the same path _independently_ from you, etc.)

For me that is bad enough to wait and see how mockito converges to a good-enough solution. I suspect it should happen pretty soon, because it appears to be a de-facto standard solution for testing in Dart.

Also, look at [a comment from Sander](https://github.com/PhilipsHue/flutter_reactive_ble/pull/176#discussion_r546597933) noting some recent progress with null-safety support in mockito.

@spkersten I'd suggest to continue the discussion here, in this ticket.

spkersten, 2020-12-21:
I wasn't aware of the whole discussion, I just tried the method with manual mock with `super.noSuchMethod(Invocation.method(#start, [...], {#name: name, ...})` and that seems to work. Still manual work of course, but maybe less than completely manual stubs. But the other issues @remonh87 mentions are valid of course.

If we don't anticipate a whole lot of new tests or interface changes, the maintenance of manually written stubs might not be so bad.

remonh87, 2020-12-21:
The main problem is that the built_value dependency of mockito is not compatible with the null-safety version of protobuf.  I am not sure how much time mockito will need to fix the issues. I would like to offer users already a version end of December begin of January to experiment since there may be some issues and it will give us some time to fix them before Google releases null-safety in stable (expected q1 2020).  

@spkersten  @werediver  In that light I can also merge my pr in the `null-safety` branch and we can always decide to merge it later in the main branch?

werediver, 2020-12-21:
> If we don't anticipate a whole lot of new tests or interface changes, the maintenance of manually written stubs might not be so bad.

I don't see this library as "completed", so I do anticipate refactoring (more of the implementation, much less of the interface) happening along with some fixes and feature development. Which implies reworking tests.

werediver, 2020-12-21:
> @spkersten @werediver In that light I can also merge my pr in the null-safety branch and we can always decide to merge it later in the main branch?

In a separate branch any experiments are fine, unless the intention is to merge that branch into the main branch eventually, which it is, right? Then we should agree on the approach to avoid wasting effort.

I haven't tried the null-safety-compatible variant of mockito and haven't read a lot about its problems, so I'd say, if both of you, @spkersten and @remonh87, are in favor of going for hand-written stubs, I trust your choice, because you have analyzed more information on the question.

spkersten, 2020-12-21:
If the public interface didn't change, I'm also fine with a null-safety release from a separate branch with hand-written stubs.

spkersten, 2020-12-21:
Another possibility is to mark the tests with `// @dart=2.9` to opt-out of null safety for them (in anticipation of good solution for mockito).

werediver, 2020-12-21:
@spkersten @remonh87 To be honest, this sounds the best to me.

spkersten, 2020-12-21:
mockito (and thus build_value) is only a dev dependency, so I think it shouldn't impose anything (the null-unsafe version) on the users of the package. So I also thing opting out the tests is a good first step.

remonh87, 2020-12-21:
If I am building the example app with the rest of the libraries upgrade to null-safety, I run into this issue: 

```
Because protobuf >=2.0.0-nullsafety.0 depends on fixnum ^1.0.0-nullsafety.0 and built_value >=5.5.5 <8.0.0-nullsafety.0 depends on fixnum ^0.10.0, protobuf >=2.0.0-nullsafety.0 is incompatible with built_value >=5.5.5 <8.0.0-nullsafety.0.
And because mockito >=4.1.2 depends on code_builder ^3.4.0 which depends on built_value ^7.0.0, protobuf >=2.0.0-nullsafety.0 is incompatible with mockito >=4.1.2.
So, because flutter_reactive_ble depends on both protobuf ^2.0.0-nullsafety.0 and mockito ^4.1.2, version solving failed.
```

I do not like excluding tests for null-safety since I  think moving to stubs instead of mocks is better for the following reason: 

- less coupled to implementation details. See also the pull request where I removed some tests that were testing implementation details rather then the actual outcome or state of the tests. I like the explanation of Martin Fowler about this: [article](https://martinfowler.com/articles/mocksArentStubs.html) .

I do agree generated stubs would be beneficial on the long run but for now excluding tests on the null-safety branch would be not something I would like to do. So I propose to migrate this solution for the null-safety branch only. Also let's keep an eye on the mockito library to see how it evolves.

remonh87, 2020-12-23:
Noticed that just mockito released null-safety version. I will check this out in the next days and see if I can get rid of the manual stubs

spkersten, 2020-12-23:
Nice timing. That would make opting-out the tests from null safety feasible. I still think that is the best short term solution as it minimizes the difference between the null-safety and main branches.

remonh87, 2020-12-23:
I agree not happy myself with these diffs.  Maybe opting out on short term is best option.

 I have a small preference to keep using generated stubs since the tests will be nicely decoupled from implementation details . Maybe I will do this first in master and then start migrating to null-safety. What do you think?

remonh87, 2020-12-23:
@spkersten  @werediver  see #177  now the mocks are generated and we can base our null-safety branch from there

werediver, 2020-12-23:
@remonh87 Please, answer my comments under #177 before proceeding with that pull-request.

remonh87, 2020-12-28:
Issue can be closed since problems are addressed in the pr and mockito has a null-safe version.

closed 2020-12-28 by remonh87