PhilipsHue/flutter_reactive_ble #175: Reading periodically from a ble device

Wassim24, 2020-12-20:
Hi everyone,

I'm working on a project implying a BLE device and an Android phone. The main idea of the project, is that every now and then, the smartphone go read some data from the BLE device, and updates the UI accordingly.

The rate at which the phone reads the BLE devices, is configurable, and may vary from 250ms to couple of minutes.

I started from the example application of flutter_reactive_ble and built the app upon it. 

My question, is how can achieve this "periodic" reading with this library ?

This is the code responsible of finding nearby devices, and and listen to their changes:
`class ExRacingScreen extends StatelessWidget {
  @override
  Widget build(BuildContext context) {
    return Consumer4<ExBleDeviceConnector, BleScannerState, ExBleDataHolder, ExDeviceDataManager>(
      builder: (_, de, scannerState, newDataHolder, deviceManager, __) {
        return WillPopScope(child: _getBody(scannerState, deviceManager, de, newDataHolder), onWillPop: () async {
          context.read<ExNavigation>().navigateBack();
          return false;
        });
      },
    );
  }

  _getBody(BleScannerState scannerState, ExDeviceDataManager deviceManager, ExBleDeviceConnector de, ExBleDataHolder newDataHolder) {
    if (scannerState.scanIsInProgress) {
      return CircularProgressIndicator();
    } else if (scannerState.discoveredDevices.isNotEmpty) {
      deviceManager.updateDiscoveredDataHolders(scannerState.discoveredDevices.map((e) => e.id).toList());
      de.listenToDevices(deviceManager.discoveredDevices);
      if(newDataHolder != null) {
        deviceManager.updateDataHolder(newDataHolder);
        return _buildNewUi(deviceManager);
      }else {
        return Text('Waiting for devices data...');
      }
    } else
      return Text('No devices found');
  }

  _buildWatchedThingStatuses(ExDeviceDataManager devicesData) {
    return ListView.separated(
      separatorBuilder: (_, index) => SizedBox(height: ExonicsTheme.smallSize),
      itemCount: devicesData.discoveredDevices.length,
      itemBuilder: (_, index) => Column(
        children: <Widget>[
          Text(
            devicesData.discoveredDevices[index].watchedThing.position.toString().split('.').last.toUpperCase(),
            style: ExonicsTheme.defaultTheme.textTheme.headline4,
          )
        ],
      ),
    );
  }
}`

And here is the implementation of my devices connector, mainly inspired from the example app:

`class ExBleDeviceConnector extends ExReactiveState<ConnectionStateUpdate> {
  ExBleDeviceConnector(this._ble);

  final FlutterReactiveBle _ble;
  final AudioCache player = AudioCache(prefix: 'assets/sounds/');
  final _deviceConnectionController = StreamController<ConnectionStateUpdate>();
  final _bleDataHolderController = StreamController<ExBleDataHolder>();

  Stream<ExBleDataHolder> get dataHolderState =>_bleDataHolderController.stream;
  StreamSubscription<ExBleDataHolder> _bleDataHolder;

  @override
  Stream<ConnectionStateUpdate> get state => _deviceConnectionController.stream;
  StreamSubscription<ConnectionStateUpdate> _connection;

  Future<void> connect(String deviceId) async {
    if (_connection != null) {
      await _connection.cancel();
    }
    _connection = _ble
        .connectToDevice(id: deviceId)
        .listen(_deviceConnectionController.add);
  }

  Future<void> disconnect(String deviceId) async {
    if (_connection != null) {
      try {
        await _connection.cancel();
      } on Exception catch (e, _) {
        print("Error disconnecting from a device: $e");
      } finally {
        // Since [_connection] subscription is terminated, the "disconnected" state cannot be received and propagated
        _deviceConnectionController.add(
          ConnectionStateUpdate(
            deviceId: deviceId,
            connectionState: DeviceConnectionState.disconnected,
            failure: null,
          ),
        );
      }
    }
  }

  listenToDevices(List<ExBleDataHolder> dataHolders) {
    dataHolders.forEach(listenToDevice);
  }

  listenToDevice(ExBleDataHolder dataHolder) async {
    if (_bleDataHolder != null) await _bleDataHolder.cancel();
    String deviceId = dataHolder.watchedThing.sensor.mac;

    _bleDataHolder = CombineLatestStream.combine2(
      readCamera(deviceId).asStream(),
      readPressure(deviceId).asStream(),
      (ExCamera camera, ExPressure tpms) {
        dataHolder.setTpms(tpms).setCamera(camera);
        return dataHolder;
      },
    ).listen(_bleDataHolderController.add);
  }

  Future<ExCamera> readCamera(String deviceId) async {
    ExCameraConstants constants = await _readCameraEeprom(deviceId);
    String data = await _readCameraData(deviceId);
    if (data.isNotEmpty) {
      return ExCamera.fromRawData(constants, data);
    }
    return ExCamera();
  }

  Future<String> _readCameraData(String deviceId) async {
    final characteristic = QualifiedCharacteristic(
      serviceId: Uuid.parse(ExBleCharacteristics.CAMERA),
      characteristicId: Uuid.parse(ExBleCharacteristics.CAMERA_DATA),
      deviceId: deviceId,
    );
    final response = await _ble.readCharacteristic(characteristic);
    if (response != null && response.length > 0) {
      return hex.encode(response);
    }
    return '';
  }

  Future<ExCameraConstants> _readCameraEeprom(String deviceId) async {
    if (ExBleCharacteristics.getCameraConstants(deviceId) == null) {
      final characteristic = QualifiedCharacteristic(
        serviceId: Uuid.parse(ExBleCharacteristics.CAMERA),
        characteristicId: Uuid.parse(ExBleCharacteristics.CAMERA_EEPROM),
        deviceId: deviceId,
      );
      final eeprom = await _ble.readCharacteristic(characteristic);
      if (eeprom != null && eeprom.length > 0) {
        ExBleCharacteristics.addDeviceCameraConstants(
            deviceId, hex.encode(eeprom));
      }
    }

    return ExBleCharacteristics.getCameraConstants(deviceId);
  }

  Future<ExPressure> readPressure(String deviceId) async {
    final characteristic = QualifiedCharacteristic(
      serviceId: Uuid.parse(ExBleCharacteristics.PRESSURE),
      characteristicId: Uuid.parse(ExBleCharacteristics.PRESSURE_DATA),
      deviceId: deviceId,
    );
    final response = await _ble.readCharacteristic(characteristic);

    if (response != null && response.length > 0) {
      return ExPressure.fromRawData(hex.encode(response));
      // fixme move this code elsewhere
      // if (!_isTiresPressureOptimal(_tpms.pressure, _tpms.temperature, dataHolder.watchedThing))
      //   player.play('beep_warning.ogg');
    }

    print('could not read pressure data');
    return ExPressure();
  }

  Future<void> dispose() async {
    await _deviceConnectionController.close();
    await _cameraController.close();
    await _pressureController.close();
    await _bleDataHolderController.close();
  }
}
`

Note: the devices are bonded with the phone during the process of adding them to the app. 

I'm really struggling with this, i was using flutter blue, and a **Timer.periodic** to read the data. Is it doable with flutter_reactive_ble ?

Thank you very much for your help guys !

remonh87, 2020-12-20:
@Wassim24  I think `Stream.periodic` is what you need [docs](https://api.dart.dev/stable/2.10.4/dart-async/Stream/Stream.periodic.html) .

you will get something like

```dart
 Stream<void>.periodic(Duration(seconds: 1)).listen((_) {
      _ble.readCharacteristic(char);
    });
```

Wassim24, 2020-12-21:
Hello @remonh87 !

Thank you very much for your help :)
I'm gonna test it right now !

closed 2020-12-22 by remonh87

Wassim24, 2020-12-24:
It worked like a charm ! 
Thank you very much :)