PhilipsHue/flutter_reactive_ble #179: How to discover characteristic type?

vaibhiarora03, 2020-12-28:
I was integrating this library for connecting and communicating with our BLE device and different third party BLE devices. The issue I am facing is that I'm unable to discover characteristic properties whether the characteristic is a write/read/notify type. 

We do not have any fixed service ids or characteristic ids to work with. Thats why we scroll across all the characteristics of a services and then individually select a write and notify characteristic and then perform the operations for connecting the device. Sometimes a single characteristic contains both write and notify properties and sometimes they are in separate characteristics. For making a trial connection and identifying whether the device is compatible with our app and functionality we perform a write request with a particular protocol and upon the response of that protocol we define the compatibility of that device and further we assign those particular set of write and notify characteristic for our further communication with device. 

Can you please help me in this? 

Thanks

remonh87, 2020-12-28:
@vaibhiarora03  in short to determine whether a device is compatible with your app it needs either: 

- Advertising a specific service that is unique for the device which you can use: 
```dart
flutterReactiveBle.scanForDevices(withServices: [serviceUuid], scanMode: ScanMode.lowLatency).listen((device) {
      //code for handling results
    }, onError: () {
      //code for handling error
    });
```

- Have specific bits in the manufacturer data that you can filter out in the discovery stream. 

See also this [link](https://www.novelbits.io/bluetooth-low-energy-advertisements-part-2/) for more info.

vaibhiarora03, 2020-12-28:
This would be limited for our devices only as we would be having the necessary data and IDs of respective characteristics. This would not help me in case connecting any third party devices. What we do is just scroll through all the characteristics of each service of the device.

werediver, 2020-12-28:
In general, Remon's advice is good: you can only connect to a new device after you receive a connectable advertisement from it, and at that point you should decide if you can interact with the device. Why would you try to communicate with an unknown device? If you can't identify the device, you don't know its protocols.

But to answer your original question, our library does not currently expose the characteristic flags (read/write/notify/indicate), so the only way to check whether the characteristic has a particular flag is to attempt to perform the corresponding operation on it and judge by the outcome. One downside of this approach is that failure codes are pretty obscure in this library, so it will be tricky to distinguish between a write failure due to non-writable characteristic or due to invalid data rejected by the peripheral, for instance.

vaibhiarora03, 2020-12-28:
Okay. Got your point. Thanks for the help.

werediver, 2020-12-28:
Created #181 to track the feature request.

closed 2020-12-28 by werediver