PhilipsHue/flutter_reactive_ble #182: Simulation Question

wolfc01, 2020-12-28:
Hi,

simple question, is there a possibility to test against a BLE simulator, like polidea is doing?

Best regards
Carl.

remonh87, 2020-12-30:
This would require some more thinking and refactoring to be honest. I am also not convinced of it. It can be useful for learning purposes but my experience is that  you would like to program against a real device to tweak the advertising and connection parameters.

wolfc01, 2020-12-31:
Hi 

Thanks for the answer. I forgot to mention the rationale for such a simulator: it could help with automatic tests of the library. Im my opinion it will help to detect issues early. It certainly will not detect everything. General rule: the earlier issues are detected the cheaper it is to fix...

happy newyear!

Br
Carl.

remonh87, 2021-01-13:
I think on the short term I won't expect a simulator due to other priorities (e.g. migration to null-safety and improving example app). I will label it as a feature that we can consider implementing at some time.

wolfc01, 2021-01-13:
Hi Remon,

yes, understood. So we'll leave this option open.
Br
Carl.

tmorone-rezi, 2021-06-11:
I second that feature request. Our apps connect to devices and have very complex UI flows. We are currently using Blemulator to simulate our device characteristics and services. This allows us to work on UI code at a much faster pace than having to use real device connections.

remonh87, 2021-08-28:
We now have implemented the federated plugin structure so technically you can write your own mocked plugin and add it  to your app. For more info see [official docs](https://flutter.dev/docs/development/packages-and-plugins/developing-packages#non-endorsed-federated-plugin). There are no plans we would build a simulator ourselves since we prefer to focus on providing the best BLE experience for mobile

closed 2021-08-28 by remonh87