PhilipsHue/flutter_reactive_ble #183: Events different onconnect ios vs android

wolfc01, 2021-01-01:
**Describe the bug**
Events on connectToAdvertisingDevice are different between iPS and Android. When connecting to peripheral device i receive the DeviceConnectionState.connecting event on Android, but not on iOS.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one peripheral using connectToAdvertisingDevice()

2. Observe the DeviceConnectionState.connecting is not received on iOS.

**Expected behavior**
The DeviceConnectionState.connecting event should also received on iOS.

**Smartphone / tablet**
 - Devices: [iPhone SE , galaxy S7]
 - OS: [iOS 14.2 Android 8.0.0]
 - Package version: [2.7.1]

**Peripheral device**
- Vendor, model: pyboard rev d
- Does it run a custom firmware / software: yes

**Additional context**
The app reproducing the problem can be found at https://github.com/wolfc01/flutter_reactive_ble_uart_example  @hash d2612b3

remonh87, 2021-01-06:
On Android I explicitly yield connecting when user triggers the connect operation. @werediver  I think iOS does something similar right?

werediver, 2021-01-07:
Upon a quick inspection, the iOS-native doesn't seem to emit `.connecting` state.

As a temporary workaround, you can assume "connecting" state right after initiating a connection.

wolfc01, 2021-01-07:
Is it an idea to remove the "connecting" event from the API, and also the "disconnecting" event, als I see no use for it, making the API a bit more simple.

Br
Carl

remonh87, 2021-02-13:
@werediver  what are we going to do with this ticket. I think connecting status on Android makes a lot because sometimes connecting takes quite a while and it is conform Android OS specs. Should we implement same on iOS or close it?

werediver, 2021-02-13:
I doubt that having an event indicating "connecting" status makes much more sense that _assuming_ the "connecting" state immediately after the method call. But I don't think it's bad to have that event either. Given that the original design implies use of that event, I'd say the iOS implementation should be adjusted to comply with the current Android implementation.

pavitter-bluestone, 2021-05-18:
I noticed the same behavior with the connectToDevice function. I'm not getting a connecting event on IOS while android is getting one.
It's with an iPad 6. Generation, running iPadOS 14.4.1

PR #315 (2021-07-29): assume connecting state on iOS
same behavior as Android

fix #183 

@werediver  I started learning Swift recently so not sure if this is what you also had in mind
closed 2021-07-29 by werediver