PhilipsHue/flutter_reactive_ble #184: Detecting sudden disconnects

realrk69, 2021-01-02:
I'm using this device to connect to an NRF device. I want to detect sudden disconnects of the device.

In the example code, there is a DeviceConnectionState that monitors the connection. I want to execute some code when the device abruptly disconnects. Where would I place this?

werediver, 2021-01-02:
The stream returned from the "connect" call will be closed when the connection is terminated, therefore you can perform your finalization work in `onDone:` parameter of `.listen(⋯)` call.

realrk69, 2021-01-04:
Thanks, this works perfectly

closed 2021-01-04 by realrk69

realrk69, 2021-01-06:
Where is this connect call in the example code? Would help to get some reference. I tried placing the onDone here, but I guess this is what it does when the connection is successful.

```
Future<void> connect(String deviceId) async {
    if (_connection != null) {
      await _connection.cancel();
    }
    _connection = _ble
        .connectToDevice(id: deviceId)
        .listen(_deviceConnectionController.add, onDone: isCalibrated(false));
  }
```

werediver, 2021-01-06:
@realrk69 You should learn to work with streams and futures in Dart, because "flutter_reactive_ble" makes heavy use of streams and futures and builds its public API using these abstractions.

You can resolve your question by referring to the official documentation on [Stream.listen(⋯)](https://api.dart.dev/stable/2.10.4/dart-async/Stream/listen.html). If its not clear, I'd suggest asking such general questions on Stack Overflow or some subreddit.

`onDone:` accepts a closure, not data.