PhilipsHue/flutter_reactive_ble #187: 'Write' does not have a timeout like 'Connect' functionality have.

mrdesai109, 2021-01-07:
Bug Description :
I don't get a response when I try to write with 'writeCharacteristicWithResponse()' after the connection is successful. This happens rarely but the bug is there for sure.

Steps to reproduce the behavior:
1. Connect to one device using `connectToDevice()`.
2. Write to that same device using 'writeCharacteristicWithResponse()'.

Expected behavior:
No response is received in the code. Kindly note this happens when I try to 'write' after getting 'connected.'

Smartphone / tablet
 - Device: Samsung Galaxy m31.
 - OS: Android 10

Additional context
The connection timeout for 'connectToDevice()' works perfectly. There should be a write timeout as well for 'writeCharacteristicWithResponse()'.

werediver, 2021-01-07:
I read the _description_ of the issue as follows: write-with-response sometimes hangs indefinitely when performed right after a connection is established. At the same time the _title_ and the _additional context_ section talks about time-out on writes.

We need to clearly understand your message: is it about the write hanging indefinitely (a potential bug) or about the time-out on writes (a feature request)?

mrdesai109, 2021-01-07:
> I read the _description_ of the issue as follows: write-with-response sometimes hangs indefinitely when performed right after a connection is established. At the same time the _title_ and the [...]

Thanks for taking my issue into consideration, my message is about time-out on 'writes' (a feature request). I believe having a timeout on writes will solve the potential bug of write hanging indefinitely.

werediver, 2021-01-07:
> I believe having a timeout on writes will solve the potential bug of write hanging indefinitely.

If there is a bug with write-with-response right after connecting, time-out would only be a band-aid / workaround, not a fix.

Are you sure the write hanging is a library issue, not your peripheral issue? This is the first time I hear about such a problem.

One thing you can try is to put a short delay after a connection is established and before the write is triggered. I'd say try something like 300 ms. Even if it works, it's not a proper fix, it's a band-aid. But it may help us to understand the issue. Only make sure you gather some statistics (if the issue happens once in 20 attempts and you only thy 5 times with no issues, that doesn't mean anything).

It can still be a peripheral issue, though.

mrdesai109, 2021-01-07:
> If there is a bug with write-with-response right after connecting, time-out would only be a band-aid / workaround, not a fix.
>
> Are you sure the write hanging is a library issue, not your peripheral [...]

Alright, I will add a delay after successful connection. I will gather the stats as well. Will get back to you with my results.

mrdesai109, 2021-01-15:
Hello, I observed that my issue was only with a particular peripheral.  After adding the delay of 200 ms I was getting more successful writes on other peripherals. So if you can add the timeout for write, a user will be notified about it irrespective of whether the peripheral is faulty or not.

remonh87, 2021-01-26:
@mrdesai109 I have to agree with the comments of @werediver  on this topic. It is quite easy to implement a timeout yourself for this, since it is a very specific usecase. For our ble lights we also created a very specific implementation around this library in our app.

closed 2021-02-03 by remonh87