PhilipsHue/flutter_reactive_ble #189: Detect raspberry

EncryptedSpirit, 2021-01-12:
For the moment the library doesn't detect raspberry pi (maybe because is from RFCOMM protocol ?)

That'll be nice to have

remonh87, 2021-01-13:
If your device is using RFCOMM it would not be detected by this library.

The aim of this library is to support devices that are communicating using the BLE protocol. There are no plans to support RFComm protocol in this library. There is another library using BLuetooth serial communication but it looks a bit outdated: [link](https://pub.dev/packages/flutter_bluetooth_serial)

closed 2021-01-13 by remonh87

werediver, 2021-01-13:
Let me elaborate a bit. This library covers a subset of iOS and Android BLE API and allows working over GATT profile ([what's GATT profile?](https://www.oreilly.com/library/view/getting-started-with/9781491900550/ch04.html)). GATT profile is implemented on top of ATT protocol and ATT protocol is implemented on top of L2CAP protocol.

Modern BLE APIs allow apps to work over L2CAP directly. It would be a cool addition to the library, but currently it doesn't allow working over L2CAP directly.

RFCOMM is another protocol implemented on top of L2CAP. Maybe it would be possible to implement RFCOMM, if we covered L2CAP APIs, but at the moment that is not possible.

Having said that, I'm pretty sure you can set up some GATT services on a Raspberry Pi. Then our library will come in handy.