PhilipsHue/flutter_reactive_ble #190: Sending file from BLE device to app

bobosette, 2021-01-12:
Hi everybody. I need to send a file .txt from a device to my app (worst case almost 2mb). The BLE device divide the file into packages. I don't know if my method is correct, but I create a loop of characteristic.write/characteristic.read telling everytime what package the device has to send.
So it writes to the characteristic the package id and then it read from the characteristic the expected package
Here's my code:

`final characteristic = QualifiedCharacteristic(
            serviceId: value.last.serviceId,
            characteristicId: Uuid.parse(
                'f0001142-0451-4000-b000-000000000000'), //value.last.characteristicIds.first,
            deviceId: deviceId);
        int beforeLoop = DateTime.now().millisecondsSinceEpoch;
        for (int i = 0; i < 20; i++) {
          int checksumId = 159 + i;
          initialValue = '9,50,100,$i,0,$checksumId,0/';

          print('stringa: ' + initialValue);

          await _ble
              .writeCharacteristicWithResponse(characteristic,
                  value: utf8.encode(initialValue))
              .then((value) async {
            await Future.delayed(Duration(milliseconds: 135))
                .then((dValue) async {
              await _ble.readCharacteristic(characteristic).then((rValue) {
                print('valore $i: ' + rValue.toString());
              });
            });
          });
        }`

It works, but is it the best solution? And in case how can I speed up the transfer (for now I have to set a delay before reading, waiting for characteristic.write to finish)?

Thank you guys

remonh87, 2021-01-13:
@bobosette  I think it really depends on the peripheral itself (for example if something goes wrong does the peripheral know which part still needs to be send). However I think the delay + read for every packet  would make everything super slow and it feels a bit over cautious.  Especially when you need to send data of a size of 2mb (ble supports a max of 20bytes per time). 

What we do for sending larger data to a peripheral is the following: 

- Slice up the data into chunks 
-  write a packet using the `writeCharacteristicWithOutResponse` operation (7x). This is a fire and forget protocol and thus the most quick way to send data but you can risk to overflow the peripheral.
- to ensure our peripheral is still responding and we do not overflow it we write the 8th packet using `writeCharacteristicWithResponse` 
-  Repeat until the file is successfully sent.

You can of course consider to execute some checks in between (maybe after each 10% of the file size) but that is up to you.

bobosette, 2021-01-13:
Thank u for the reply. When I write to the ble device I only send a string with a checksum and the package id, the ble device sends the real data to the app. And you're right, the problems are the delay + read, but I had to put that delay cause if I write without response and then I go to read i receive an exception. It's like there is a delay due to wait for the ble device to receive the string and prepare the package to be send. Is it possible?

A solution that maybe I found is to use listen() instead of read so when data changes the ble device send it to the app, but in this way I have another issue: when I read i receive 64byte package but when I listen() only 20byte. Why??

remonh87, 2021-01-13:
another thing what you have to do is increase the MTU size. For android the max size is 512 (however make sure the device supports it). For got to mention this in the previous response I think this will already help a lot.

bobosette, 2021-01-13:
Sorry can you explain a little bit more about the MTU size?

Taym95, 2021-01-13:
Check these documents to understand what's MTU(Maximum Transmission Unit):
[Maximum transmission unit](https://en.wikipedia.org/wiki/Maximum_transmission_unit)

[One minute to understand BLE MTU data package](https://github.com/greatscottgadgets/ubertooth/wiki/One-minute-to-understand-BLE-MTU-data-package)

The library allows you to Negotiate MTU size using [requestMtu](https://github.com/PhilipsHue/flutter_reactive_ble#negotiate-mtu-size)

bobosette, 2021-01-13:
Ok thank you but if I want to send data from the peripheral I need to negotiate MTU size from inside the peripheral, not from the app side. Is it correct?

remonh87, 2021-01-17:
@bobosette the mtu is an agreement between the peripheral and the client. It applies to both directions.

bobosette, 2021-01-17:
Ok. So for example if I set an mtu of 64 but when I check the size is always 23, does it mean that for my peripheral the maximum size is 23?

remonh87, 2021-01-17:
yes I would expect that because on iOS the max mtu is generally 185 and on android 517. Transferring 2mb with an mtu of 23 would be super slow in my opinion and I would check with the vendor of the device if you can increase it more.

bobosette, 2021-01-17:
For sure. But the weird thing is that when i use readcharacteristic i riceive 64 byte packages, and when I use listen only 20!!

werediver, 2021-01-17:
It's not weird. Automatic fragmentation is happening on reads and writes, to some extent. It's not supported for notifications or indications.

bobosette, 2021-01-17:
Mmm ok. So, for you which is the best/correct way to get this file from a BLE peripheral? 
Cause, at this point, I'm thinking this isn't the best way. As I say in the first comment for now the peripheral splits the file in some packages, and it has a custom service with a characteristic. When it receives (with writecharacteristic) a string with the number of package to send, after readcharacteristic (or listen) it sends the package.
But I don't know, maybe another solution could be to create as many characteristic as the packages number and have a loop of readcharacteristic???

werediver, 2021-01-18:
I don't think you can easily and/or reliably make the many-characteristic approach work. It is definitely not something that is normally done.

Splitting your payload into small packets and sequentially sending them via a single characteristic is the standard approach. Often the control point abstraction is used (a characteristic with write & notify permissions, requests to a peripheral are _written_ and responses are delivered back as notifications).

The only more efficient way of transferring data over BLE that I can think of would be over the stream-oriented L2CAP API, but this library does not cover it.

closed 2021-01-31 by remonh87