PhilipsHue/flutter_reactive_ble #195: ScanFailure on android

carlos-labrador, 2021-01-18:
**Describe the bug**
After integrating FlutterFire package into our project , we are getting the following exception

`REACTIVE_BLE: Received ScanResult(result: Result<DiscoveredDevice, GenericFailure<ScanFailure>>.failure(GenericFailure<ScanFailure>(code: ScanFailure.unknown, message: "Scanning parameters are not set")))
I/flutter (14507): Exception: GenericFailure<ScanFailure>(code: ScanFailure.unknown, message: "Scanning parameters are not set")`
<details><summary>Pub Specs:</summary>
```
dependencies:
  flutter:
    sdk: flutter
  firebase_core: "^0.5.3"
  flutter_localizations:
    sdk: flutter
  flutter_reactive_ble: ^2.7.1
  firebase_messaging: "^8.0.0-dev.10"
```
</details>

**To Reproduce**
Steps to reproduce the behavior:
1. Initialize FlutterReactiveBle
2. if it is ready shows Bluetooth Scan page.
3. Users tap on scan button.

**Expected behavior**
it should show all available Bluetooth devices.

**Smartphone*
 - Device: [Samsung s7 edge]
 - OS: [Android 8]
 - Package version: [^2.7.1]

**Peripheral device**
- Vendor, model: [SC-02H]
- Does it run a custom firmware / software: n/a

**Additional context**
I could reproduce it on the base example app adding flutter fire package.

Taym95, 2021-01-18:
Hi, I think the error `Scanning parameters are not set` tells a lot, you should provide params to `scanForDevices` function, [withServices param is required](https://github.com/PhilipsHue/flutter_reactive_ble/blob/a9e4fe91eea405458ff003bcf0d9fcdbc7a9a2ee/lib/src/device_scanner.dart#L19), check [startScan](https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/example/lib/src/ui/device_list.dart#L70) in the example app.

carlos-labrador, 2021-01-18:
Hi @Taym95 , 

In my ticket says "I could reproduce it in the example app" that's mean I took the time of review it a lot.

Should you need assistance to reproduce it I kindly can tell you.

Thanks.

Taym95, 2021-01-18:
Hi @carlos-labrador, I tried to reproduce it, I added `await Firebase.initializeApp();` [here](https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/example/lib/main.dart#L14) before building UI, and I managed to scan for devices, can you share the way you used firebase?

carlos-labrador, 2021-01-19:
Hi @Taym95 ,
I already updated the example.
Let me know if you need anything.

Thanks for your support.

werediver, 2021-01-19:
Can anybody comment on whether the same is happening on iOS?

jjchiw, 2021-01-19:
Hi

Yes it works fine in iOS

remonh87, 2021-01-20:
Hmm looks like some race condition in combination with that method. I will take a look at it the coming days, already fixed something similar in the past but apparently the fix was not enough.

carlos-labrador, 2021-01-21:
@remonh87 Thanks for your feedback and hard work.

closed 2021-01-25 by remonh87

remonh87, 2021-01-25:
This issue should be fixed in version 2.7.2 . Thanks for the clear reproduction steps, was easy to reproduce it.

carlos-labrador, 2021-01-26:
@remonh87 , Thanks for your support.

riccardocescon, 2022-12-06:
@remonh87 The same thing is happening with the latest version 5.0.2.
The BLE was working perfectly, but last week i implemented the FirebaseMessage.onBackgroundMessage callback and started having this issue.
Is it possible that this issue has been  regressed with the latest versions?
Tested with:
Pixel 6 - Android 13
Xiaomi MI 9 - Android 11

insertjokehere, 2023-12-12:
@remonh87 we have also been seeing this error in production - any suggestions for how we could debug this?

AndreaBalde, 2024-03-08:
@remonh87 even with version 5.3.1 I have the same problem.
Tested with:
Pixel 7 - Android 14

chplalex, 2024-06-02:
@remonh87 the same issue with 5.2.0
Tested with Android 12.
iOS works fine.

PR #200 (2021-01-25): fix reinitializing of the eventchannel scanparameter by firebase
There is nothing wrong with the previous code however for some weird reason firebase resets state of the eventchannels which looks really fishy. Also there are quite some bugs there. This workaround however makes sense and guarantees good functioning state.

Fixes #195