PhilipsHue/flutter_reactive_ble #197: Background reconnect

katilley, 2021-01-18:
First, nice work on this package. Simple to use and reliable so far. Thanks!

I'd like my app to handle automatically reconnecting after a connection has been lost when the app is in the background. Any pointer or guidance appreciated. I've been looking at isolates and the workmanager package but want to make sure I start down the right path or find an example if one exists.

remonh87, 2021-01-18:
@katilley  good to hear that our library works. It is possible to run Flutter code in the background. This [article](https://medium.com/flutter/executing-dart-in-the-background-with-flutter-plugins-and-geofencing-2b3e40a1a124) is what I think you are looking for. It is I think even part of the Flutter official docs now. Instead of using a foreground service feel free to use any background processing you like. Hope this helps.

katilley, 2021-01-20:
Thanks! I've looked through the geofencing example several times but it is so specific to platform supported geofencing it isn't obvious to me how to really leverage it. I'll keep looking.

As an example, it says "... there is no guarantee that the background isolate will stay alive while the application itself is backgrounded." I guess his example gets around this because he's using platform supported geofencing. 

He's also implementing a lot of platform specific code which i'd like to avoid if possible.

remonh87, 2021-01-20:
@katilley the example is specific to geofence however it describes exactly how to do dart background processing. I would study that part and ignore all the geofence specific stuff.  In short what you need to do:
- create some machinery to spawn up a dart isolate on the android background thread
-  a method in dart where you execute the ble specific code (in the article called the appcallback)
- Create a very basic plugin where you save the dart appcallbackid (in for example sharedprefs)
- Some native plumbing where you execute the specific callback

As far as I know this is the only way to do background processing in Flutter.

closed 2021-01-30 by katilley