PhilipsHue/flutter_reactive_ble #198: service data returning null

Virendra0014, 2021-01-19:
Hello there,

Using the example provided by this library, I had connected to a ble device successfully but while I'm trying to read ble device service data with device.serviceData always returning null. I had tried with different devices but I'm getting null every time. Actually I'm trying to get serviceID and characteristicID by using device that is connected. Is it the right way I am doing or any other way to do it please help me.

remonh87, 2021-01-20:
@Virendra0014  the service data contains the service uuid and corresponding data. What you need is  the `discoverServices ` method which provides you a list of services and corresponding characteristics. I keep this issue open since we forgot to document this in the readme .

Virendra0014, 2021-01-21:
@remonh87 I will try using the discoverServices method to fetch all the services and corresponding characteristics of a connected device .

Thanks for your help . 👍

remonh87, 2021-05-02:
this issue is now solved in version 3.1.0.

closed 2021-05-02 by remonh87