PhilipsHue/flutter_reactive_ble #199: "setCharacteristicNotification()" debug messages on LogLevel.none

d-wolf, 2021-01-22:
**Describe the bug**
I do activate and deactivate characteristic notifications periodically. At the same time, the LogLevel of FlutterReactiveBle is set to none (default). Nevertheless i get debug messages of  setCharacteristicNotification() like ```D/BluetoothGatt(31815): setCharacteristicNotification() - uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e enable: true``` & ```D/BluetoothGatt(31815): setCharacteristicNotification() - uuid: 6e400002-b5a3-f393-e0a9-e50e24dcca9e enable: false```. I would expect that there is no debug output if LogLevel is set to none. Because of my special use case my debug console gets flooded with these messages.

**To Reproduce**
Steps to reproduce the behavior:
1. connect to one device with ```_ble.connectToDevice(id: peripheralId, connectionTimeout: timeout)```
2. subscribe a characteristic```  _notifySubscription = _ble.subscribeToCharacteristic(characteristic).listen(_handleIncomingData, cancelOnError: true)```
3. unsubscribe the characteristic ```await _notifySubscription?.cancel()```

**Expected behavior**
No debug messages should be shown on LogLevel none.

remonh87, 2021-01-24:
This debug logging from android. There is nothing what we can do about it to prevent it. If I am correct this message will not be displayed when the app is in release mode.

closed 2021-01-24 by remonh87

d-wolf, 2021-01-25:
Okay, thanks for your help @remonh87 .