PhilipsHue/flutter_reactive_ble #201: W/FlutterJNI(13077): Tried to send a platform message to Flutter, but FlutterJNI was detached from native C++. Could not send.

antigones, 2021-01-25:
**Describe the bug**
An app that subscribe to a characteristic, connecting to an Arduino Nano 33 BLE used as thermometer.
When the app goes in background, I sometimes received the message:

W/FlutterJNI(13077): Tried to send a platform message to Flutter, but FlutterJNI was detached from native C++. Could not send. 

and app is not able to reconnect when returning to foreground.

**To Reproduce**
Steps to reproduce the behavior:

1. Download sources from:

https://github.com/antigones/flutter_arduino_nano_33_ble/tree/subscription/flutter

2. Send app to background
3. When returning to foreground, app can't re-establish a connection and:

W/FlutterJNI(13077): Tried to send a platform message to Flutter, but FlutterJNI was detached from native C++. Could not send. 

pop out in the terminal.

**Expected behavior**
App should be able to reconnect.

**Smartphone / tablet**
 - Device: Honor View 10
 - OS: Android 10
 - Package version: 2.7.2

**Peripheral device**
- Vendor, model: Arduino Nano 33 BLE Sense
- Does it run a custom firmware / software: no

remonh87, 2021-01-26:
@antigones I did have a look to the code that you supplied and I would suggest a couple of changes. 

You initialize the `FlutterReactiveBle` Library in the state object. Can you initialize this in the main and inject it? I think the main issue is related in there.  Within Philips Hue we automatically reconnect when coming from the background without issues using my suggested approach.

antigones, 2021-01-27:
@remonh87 I tried to do as suggested but no luck (probably there is something that I am missing here).
For reference, I updated the code in the repository.

This issue comes specifically up when I pop the thermometer application, open other apps, go around the system, do stuffs and go back to thermomether app. At that point I see a blank white screen because the app is not able to recover in order to disconnect and reconnect again. I am trying to replicate the issue on a stable basis but it seems totally random.

I think I will try to intercept the back button to disconnect before the app is sent to background.

remonh87, 2021-01-27:
I am still not able to reproduce it on the example app. Can you try using our example app and connect with the ble peripheral and see if the issue still occurs? It can be even a very specific issue which makes it even harder to reproduce it.  

Also maybe you can try in your app to implement a timeout or catching the error (if any error is thrown) and try it again.

antigones, 2021-01-28:
I tried and the example app is totally ok, I can navigate back and forth between apps with no problem even using the hardware button, so the issue is totally residing in my code. :)

Thank you for the suggestion, I will try to go deeper into the problem!

remonh87, 2021-01-28:
Ok good luck with that, if you have more questions feel free to let us know

remonh87, 2021-02-03:
Will close it for now feel free to reopen or file another issue in case you run into some issues.

closed 2021-02-03 by remonh87

huuang, 2021-03-10:
@antigones 

I ran into the same problem, how did you fix it?

antigones, 2021-03-14:
@huuang sorry for the delay, this source seems to work for me:

https://github.com/antigones/flutter_arduino_nano_33_ble/tree/subscription

Let me know if you incur in the same issue.

iBog, 2022-02-02:
Found the same issue when trying to close BLE connect in dispose() method, same result for closing in didChangeAppLifecycleState (paused and detached states).
The only case when the app closed fine is to stop BLE in onWillPop as  'subscription' branch example did.

dhaval-k-simformsolutions, 2022-03-09:
I am also facing the same issue. and it's totally random.

kakeung, 2022-07-18:
I encounter this issue every time.

**Problem to repeat**
1. Open app
2. Start scan
3. press android 'back' button to close app
4. Open app
5. Start scan again
6. You will see the error in log (Channel: flutter_reactive_ble_scan)

If you repeat the step again and again, the error log increase and the 'Response ID' is different. The problem will occurred if you close android activity and flutter view, but no really kill the app process.

I think the problem is that when we restart the flutter engine and invoke the `onAttachedToEngine`. It recreate the `ReactiveBlePlugin` object. That means some object not fully deallocate or dispose in Android native when flutter invoke `onDetachedFromEngine`?

flutter_reactive_ble: v5.0.2
OS: Android 10/11

splangi, 2022-07-26:
> I encounter this issue every time.
>
> **Problem to repeat**
>
> 1. Open app
> 2. Start scan
> 3. press android 'back' button to close app
> 4. Open app
> 5. Start scan again
> 6. You will see the error in log [...]

I would add a step 0 to reproduce for everyone: Turn on "Don't keep activities" from developer options. 

Funnily enough there is a fix commited but it is unpublished! 

Adding this can use the commited fix (Bevare in production! This uses master branch):

`dependency_overrides:
  reactive_ble_mobile:
    git:
      url: git@github.com:PhilipsHue/flutter_reactive_ble.git
      path: packages/reactive_ble_mobile`