PhilipsHue/flutter_reactive_ble #202: Build failed with an exception when added to dependencies

JKornberg, 2021-01-26:
**Describe the bug**
Unable to build my project on an Android emulator when adding `flutter_reactive_ble: ^2.7.2` to the pubspec.yaml.

**To Reproduce**
Steps to reproduce the behavior:
1. Add flutter_reactive_ble: ^2.7.2
2. Run project on Android Emulator in debug mode on Visual Studio Code

**Expected behavior**
App should run properly

- [ ] I tried making a fresh project and adding it to pubspec.yaml and same thing happened

**Smartphone / tablet**
 - Device: Pixel 4 XL API 30 Emulator
 - OS: Android
 - Package version: ^2.7.2

**Additional context**
Am I not allowed to run this app on emulators? I'm sorry if this is documented somewhere but I could not find it.

denkoul, 2021-01-27:
same problem here

remonh87, 2021-01-27:
@JKornberg  can you send me the output of `flutter doctor`  and also provide me the version of Android studio + gradle wrapper and the android-tools version you are using?   The app builds properly on my machine on flutter stable and beta with stable Android studio.

denkoul, 2021-01-28:
With the following configuration, the plugin is working:
`com.android.tools.build:gradle:4.0.1` in android/build.gradle
`distributionUrl=https\://services.gradle.org/distributions/gradle-6.1.1-all.zip` in gradle wrapper properties

When I update gradle to 4.1.0 and distr url set to gradle-6.5 (required to run ndk.debugSymbolLevel) , the plug in crashes with the following error:
`A problem occurred configuring project ':flutter_reactive_ble'.
> Failed to notify project evaluation listener.
   > com.android.build.gradle.internal.variant.BaseVariantData.getScope()Lcom/android/build/gradle/internal/scope/VariantScope;`

remonh87, 2021-01-28:
Hmm this looks likes a bug will investigate this in the coming days. Probably we need to update some android native dependencies on our site.

remonh87, 2021-02-02:
Issue is resolved in version: `2.7.3`

PR #210 (2021-02-02): upgrade android dependencies to comply to newer gradle distributions
Fix #202
closed 2021-02-02 by remonh87