PhilipsHue/flutter_reactive_ble #203: Service Discovery Failure

realrk69, 2021-01-28:
I'm using the method connect to add some configuration for the device which I am connecting to:

```
Future<List<List<int>>> connect(String deviceId) async {
    if (_connection != null) {
      await _connection.cancel();
    }
    _connection = _ble
        .connectToDevice(id: deviceId)
        .listen(_deviceConnectionController.add, onDone: () {});
    await _ble.discoverServices(deviceId).then(
          (value) => print('Services discovered: $value'),
        );

    _tx = QualifiedCharacteristic(
        serviceId: deviceUUID, characteristicId: writeUUID, deviceId: deviceId);
    _rx = QualifiedCharacteristic(
        serviceId: deviceUUID, characteristicId: readUUID, deviceId: deviceId);
    _rxSubscription = _ble.subscribeToCharacteristic(_rx).listen((event) {
      _response.add(event);
    });
    print("success");
    print(deviceId);
    // Write calibration settings
    final _calibrateResponse = await writeCalibrationSettings();
    if (_calibrateResponse != null)
      return _calibrateResponse;
    else
      return null;
  }
```
Calling it:
```
onTap: () {
    try {
        final calibrateResponse = await connect(device.id);
        if (calibrateResponse == null)
            print('Error Occurred')
    }
    catch (e) {
        print(e);
    }
}
```

The error:
```
PlatformException(service_discovery_failure, null, null, null)
```

I don't understand why this started happening suddenly. I tried reinstalling the app, restarting the phone, disconnecting and reconnecting to the device, and powering the device off for 30 seconds. If I connect to another smartphone, it works and then suddenly stops after some time.

remonh87, 2021-02-02:
@realrk69  this exception normally happens in case the a platform interaction failed. The reason can be anything. Can you provide a bit more elaborate logs since it will be hard for me to judge the issue based on only this error.  Probably there will be some crash or error before this one that can give more context.

realrk69, 2021-02-10:
Turns out, my device had an error and it was not related to the app or the package at all. Its just that the error occurs sometimes, which is quite weird. Thanks for the advice @remonh87

closed 2021-02-10 by realrk69