PhilipsHue/flutter_reactive_ble #204: need a callback when pairing with code

denkoul, 2021-01-28:
When a sub to a characteristic needs pairing, a callback notifying that the pairing was successful would be nice.

werediver, 2021-01-28:
Any BLE operation will only complete successfully when the user allows pairing. At least on iOS no callbacks can be implemented for that. _Imagine pairing doesn't exist_, this is how iOS presents it to the app. An operation succeeds, if the pairing succeeds (and the operation itself, of course), and fails, if the pairing fails.

@remonh87 Could you comment about the Android side?

denkoul, 2021-01-28:
Ok, @werediver , I think you got me covered

remonh87, 2021-01-28:
@werediver  @denkoul  in Android you can request the paired devices, so in theory we can do sth like that. However, like @werediver  explained. if pairing with the device fails the read characteristic will provide an error .

closed 2021-01-31 by remonh87