PhilipsHue/flutter_reactive_ble #205: iOS When I Turn off BLE my app crashes

lwinged, 2021-01-29:
Hello @werediver @remonh87 ,

when I turn off the Bluetooth after subscribed to characteristics.

I subscribe to characteristics (it works) when device is connected and when I turn off the Bluetooth I have a crash.

When I compile with Xcode in debug mode the app doesn't crash the console prints the error.  But with VSCode I have a crash (cf screens) but I can go next (with breakpoint bar) and the app goes on to work.

Thanks for your help.

crash/error:
```
flutter: connectToDevice ConnectionStateUpdate(deviceId: D80B4858-9E93-29E8-06B2-6A7F9A74DCE6, connectionState: DeviceConnectionState.disconnected, error: GenericFailure<ConnectionError>(code: ConnectionError.failedToConnect, message: "Bluetooth is not powered on (the current state code is 4)"))
flutter: _monitor: BleStatus.poweredOff
flutter: Error unsubscribing from notifications: PlatformException(flutter_reactive_ble.Central.(unknown context at $1039c2e98).Failure:1, The operation couldn’t be completed. (flutter_reactive_ble.Central.(unknown context at $1039c2e98).Failure error 1.), {}, null)

```

example of code : 

```
class _BlePageState extends State<BlePage> {
  StreamSubscription monitorSub;
  StreamSubscription scannerSub;
  StreamSubscription connectionSub;
  StreamSubscription charSub;
  FlutterReactiveBle ble;

  @override
  void initState() {
    super.initState();
    ble = FlutterReactiveBle();
    _monitor();
  }

  void _monitor() {
    monitorSub = ble.statusStream.listen((state) async {
      print("_monitor: $state");
      if (state == BleStatus.ready) {
        _scan();
      } else {
        await charSub?.cancel();
        await connectionSub?.cancel();
        await scannerSub?.cancel();
      }
    },
        onError: (error) => print("monitor On error $error"),
        onDone: () => print("monitor on done"));
  }

  void _scan() {
    scannerSub = ble
        .scanForDevices(
            withServices: [kBTService].map((e) => Uuid.parse(e)).toList())
        .where((device) => device.name == kCorailHardwareName)
        .take(1)
        .listen((device) async {
      print("scan: $device");
      await scannerSub.cancel();
      connectionSub = ble.connectToDevice(id: device.id).listen((state) async {
        print("connectToDevice $state");
        if (state.connectionState == DeviceConnectionState.connected) {
          final characteristic = QualifiedCharacteristic(
              serviceId: Uuid.parse(kBTService),
              characteristicId: Uuid.parse(kBTCorailUICharacteristic),
              deviceId: device.id);

          charSub =
              ble.subscribeToCharacteristic(characteristic).listen((values) {
            print(values);
          });
        }
      });
    }, onDone: () {
      print("Scan Done");
    });
  }

  @override
  void dispose() {
    monitorSub?.cancel();
    super.dispose();
  }
```

Flutter: 1.22.5
Package version: 2.7.2 and 2.4.0+1
iOS 13.4.1
iPhone 6s Plus

[image]
[image]
[image]
[image]

_Originally posted by @lwinged in https://github.com/PhilipsHue/flutter_reactive_ble/issues/137#issuecomment-769302636_

remonh87, 2021-02-02:
@werediver  can you comment on this issue? For me this looks like some race condition

lwinged, 2021-02-02:
@remonh87 @werediver 

Just to precise : I put Future.delay (5secs) before each call but I had the same result.

werediver, 2021-02-02:
I see no issues here. A connection gets terminated abnormally terminated (rather than intentionally closed), which results in reporting an exceptional situation (throwing an exception).

@remonh87 Do you mean the Android part handles this situation in a different way?

remonh87, 2021-02-02:
@werediver  in this case android would also throw an exception. The only thing what I can think of that explains the difference is the timing on the IPC channels, but behavior is implemented the same.

werediver, 2021-02-03:
@remonh87 Could you explain, what difference do you mean? I said that an abrupt connection termination leads to exceptions on all operations in progress. That is expected. The client should be ready to handle IO failures, regardless of which IO it performs.

werediver, 2021-02-09:
For about a week there is no activity here. As I stated before, I consider the described behavior _correct_, thus closing the issue.

Feel free to reopen, if needed.

closed 2021-02-09 by werediver