PhilipsHue/flutter_reactive_ble #207: Bluetooth connection on Android issue

carlos-labrador, 2021-02-01:
Describe the bug:
Following up with the issue #195 , it seems ConnectionStateUpdate stream does not change.

Pub Specs:
To Reproduce
Steps to reproduce the behavior:

Initialize FlutterReactiveBle
tap on scan button.
select a Bluetooth device from the list.
tap on connect button.

Expected behavior
it should show status: DeviceConnectionState.connecting or DeviceConnectionState.connected

*Smartphone

- Device: [Samsung s7 edge]
- OS: [Android 8]
- Package version: [^2.7.1]
- Peripheral device

Vendor, model: [SC-02H]
Does it run a custom firmware / software: n/a

**Additional context**
I could reproduce it on the base example app adding flutter fire package.

AayuHub, 2021-02-02:
Hi!
I am having a similar issue. 
Is there any way to make this work?

remonh87, 2021-02-03:
I hope I will have the upcoming days some time to work on the bugs (currently very busy). It is on my radar :-)

nifrz, 2021-02-11:
I have the same problem without Firebase messaging. It seems that after calling `connectToDevice` the method `listen` is not called, because if I set a breakpoint at the beginning of `listen` the debugger does not reach it.
But the connection is established!

I have updated to version `^2.7.3`, but the problem still exists in version `^2.7.1`.

Thank you very much for your work! I hope there will be a solution soon, because I need this functionality.

remonh87, 2021-02-13:
I cannot reproduce the issue. I installed the following depedencies in pubspec.yaml in the example app and initialized firebase. I am able to connect and do operariont on the app without problems.

```yaml
  flutter_reactive_ble:
    path: ../.   
  firebase_messaging: ^7.0.3
  firebase_core: ^0.5.3  
```

@carlos-labrador I noticed you use a null safety version of firebase-messaging which can be one of the causes.

closed 2021-02-13 by remonh87