PhilipsHue/flutter_reactive_ble #208: connect the same device with the same app on different phone at the same time

CyJaySong, 2021-02-01:
**Describe the bug**
Hello, I try to run the same app on different phone at the same time to connect the same device, but only the first connected device can be successfully connected. 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one / many devices using `connectTo()` / `connectToAdvertisingDevice()`
2. Observe a failure with exception (including the part of the stack trace, belonging to this package) …

**Expected behavior**
- [ ] Support to use the same app to connect the same device in multiple phone at the same time

**Smartphone / tablet**
 - Device: [iPhone 7p,Mi8]
 - OS: [iOS 14, Android 10]
 - Package version: [2.7.2]

**Peripheral device**
- Vendor, model: DL-581PW
- Does it run a custom firmware / software:  no

**Additional context**
-

werediver, 2021-02-01:
This library has no limitations that can relate to your issue. Please look into your peripherals capabilities. The firmware may be written so that it does not handle multiple connections.

closed 2021-02-01 by werediver

CyJaySong, 2021-02-02:
> 该库没有与您的问题有关的限制。请查看外围设备的功能。固件可能被编写为不处理多个连接。

> This library has no limitations that can relate to your issue. Please look into your peripheral's capabilities. The firmware may be written so that it does not handle multiple connections.

bug, I try to connect with other Bluetooth applications, which is feasible.

CyJaySong, 2021-02-02:
> This library has no limitations that can relate to your issue. Please look into your peripheral's capabilities. The firmware may be written so that it does not handle multiple connections.

The peripheral device supports three IOS apps and six Android apps

werediver, 2021-02-02:
Provide the details on the failure you're observing when the second device attempts to connect to your peripheral (exception, stack-trace).

@remonh87 Do you have any ideas whatsoever how this can happen due to our library? I have strong suspicion the problem is not in the library, but it's difficult to guarantee that.

@CyJaySong Just so that you know, this library is used in Hue Bluetooth app, which allows to control multiple (up to about 10, depending on the device) BLE-enabled Philips Hue lights. So even if there is an issue, it must be something quite specific.

CyJaySong, 2021-02-02:
> Provide the details on the failure you're observing when the second device attempts to connect to your peripheral (exception, stack-trace).
>
> @remonh87 Do you have any ideas whatsoever how this can [...]

I've learned some GATT related rules. ble can't connect multiple central devices to the same peripheral device at the same time.

CyJaySong, 2021-02-02:
@werediver Can you recommend a classic Bluetooth plug-in that supports multiple device connections?

werediver, 2021-02-02:
> I've learned some GATT related rules. ble can't connect multiple central devices to the same peripheral device at the same time.

This is not true. It depends on whether peripheral's firmware is capable of handling multiple simultaneous connections. It could be the case that your particular peripheral only allows a single connection, though. That would kind of make sense for a small thermal printer, I'd say.

As a matter of fact, this library is used in Hue Bluetooth app, which allows to control BLE-enabled Philips Hue lights and those lights are capable of accepting multiple connections simultaneously.

werediver, 2021-02-02:
> @werediver Can you recommend a classic Bluetooth plugin that supports multiple device connections?

You will have difficulties implementing Bluetooth Classic communications on iOS, it requires some special approvals from Apple. [CoreBluetooth](https://developer.apple.com/documentation/corebluetooth) framework, that allows Bluetooth communications on iOS, only supports BLE.

I am not sure about the situation with Bluetooth Classic on Android.

CyJaySong, 2021-02-02:
@werediver I use the classic Bluetooth method to connect my thermal printer, which can connect multiple devices at the same time

CyJaySong, 2021-02-02:
@werediver https://github.com/andrey-ushakov/flutter_bluetooth_basic 
I use this plugin to test. It can realize multiple connections on Android platform, but it can't close a single connection. This is a pity