PhilipsHue/flutter_reactive_ble #209: Add option for discoverServices to return long UUID (128 bit)

Papajik, 2021-02-01:
**Is your feature request related to a problem?**
Helo, I am working on ESP32 MCU and with custom services and characteristics using long UUID format. I want my android app to be a dynamic and show different widgets based on advertised characteristics. 

I have defined custom services on ESP32 like this: 
```
#define SERVICE_CREDENTIALS_UUID "56600000-BE5D-4370-877B-C4A2ACE639E8"
#define SERVICE_STATE_UUID "B4D00000-DFAE-4F0B-9EEF-AEA76041E2D9"
#define SERVICE_SETTINGS_UUID "7D700000-F94E-4D34-B701-483E29C79F55"
```
and example of characteristics: 
```
#define CHARACTERISTIC_UUID_PH_MAX "50A00001-4E12-11EB-AE93-0242AC130002"
#define CHARACTERISTIC_UUID_PH_MIN "50A00002-4E12-11EB-AE93-0242AC130002"
#define CHARACTERISTIC_UUID_PH_CUR "50A00003-4E12-11EB-AE93-0242AC130002"
```

When i am using another app to scan BLE device (like BLE Scanner) I am able to see long UUIDs and two mandatory services 0x1800 (Generic access) and 0x1801 (Generic attribute) which are only 16 bits. But when I try to use discoverServices in my app, all UUIDs are in short (16 bit) version:

```
I/flutter (30058): Service = 1800 ([24, 0])
I/flutter (30058): 
I/flutter (30058): Service = 1801 ([24, 1])
I/flutter (30058): 
I/flutter (30058): Service = 0000 ([0, 0])
I/flutter (30058): 
I/flutter (30058): Service = 0000 ([0, 0])
I/flutter (30058): 
I/flutter (30058): Service = 0000 ([0, 0])
```
Same goes for characteristics: 

```
I/flutter (30058): 	0001 ([0, 1])
I/flutter (30058): 	0002 ([0, 2])
I/flutter (30058): 	0003 ([0, 3])
```

When I create Service UUID in app, it uses full 128 UUID:

```
final Uuid SERVICE_UUID = Uuid.parse('7D700000-F94E-4D34-B701-483E29C79F55');
print('looking for = ' + SERVICE_UUID.toString());
print('looking for = ' + SERVICE_UUID.data.toString());

/// Output
I/flutter ( 3861): looking for = 7d700000-f94e-4d34-b701-483e29c79f55
I/flutter ( 3861): looking for = [125, 112, 0, 0, 249, 78, 77, 52, 183, 1, 72, 62, 41, 199, 159, 85]
```

**Describe the solution you'd like**
I would love to be able enforce discoverServices to return long UUIDs when service/characteristic uses 128 bit UUID and short UUID when there is short UUID advertised. Maybe using an optional parameter:
`Future<List<DiscoveredService>> discoverServices(String deviceId, {bool useLongUuids})=> ...`

If this would take too much work, is it possible to at least add description that discoverServices returns only 16 bit UUID? 

**Describe alternatives you've considered**
Since I can edit UUIDs on ESP32 I will just make sure that all 3rd and 4th bytes of UUID are unique and I will work with theirs abbreviations on the android side. But that solution is not possible when working with device where I can't change its UUIDs.

soyangel, 2021-02-01:
Was going to report this same behaviour and also using ESP32.

werediver, 2021-02-01:
That's a surprising issue. For sure the library reports long IDs eh... normally. I suspect it's specific to ESP32's BLE stack.

But because a generic BLE scanner on the same central device shows the IDs inn full, there must be something we can do (if we're lucky; if we're not lucky, it may be in [Polidea/RxAndroidBle](https://github.com/Polidea/RxAndroidBle)).

Before we poke the Android expert, please let me know whether you're experiencing the same issue when running you app on an iOS smartphone (or something).

soyangel, 2021-02-02:
Seems like on iOS long UUIDs are ok:

DiscoveredService(serviceId: 00dac273-80b3-475c-83fc-56540d498c68, characteristicIds: [01dac273-80b3-475c-83fc-56540d498c68, 02dac273-80b3-475c-83fc-56540d498c68], includedServices: [])

werediver, 2021-02-02:
Okay, then we have to summon @remonh87!

Remon, do you have any ideas how the Android native part can mangle the IDs? I don't think we experienced problems like this before... but also we haven't used service discovery in production. It's one of the features we introduced for the community.

Klievan, 2021-02-02:
FYI,, I observe the same using the nRF52 module

Papajik, 2021-02-02:
> That's a surprising issue. For sure the library reports long IDs eh... normally. I suspect it's specific to ESP32's BLE stack.

I used [NimBLE](https://github.com/h2zero/NimBLE-Arduino) so after your comment I also tried to use original [BlueDroid](https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE) library with the same flutter app and outcome is the same - serviceId and its characteristicIds are returned in 16 bit format.

werediver, 2021-02-02:
Could you research whether you can influence this from the peripheral side? Even if there is something that can be done on the central side, researching the peripheral side may help us understand what to look for.

Klievan, 2021-02-02:
I tested with another central device, the BGM121 from Silabs and I observe the same thing. I have a hunch this is a general issue on the android side. I will investigate further

remonh87, 2021-02-02:
I did some research and this is a very weird issue. In our code we did nothing wrong as far as I can see (I added some extra unittests for 128bit uuids and it works as expected).  Also I wouldn't  expect an issue on android since I notice they use 128bits https://developer.android.com/reference/java/util/UUID .  I came across some fora that some phones below android 8 have some problems with longer uuids. Just to check did you experience the same issue on different phones (also newer os versions).

soyangel, 2021-02-02:
In my case I'm seeing it on a Pixel 3a with Android 11.

Klievan, 2021-02-02:
I'm on a One Plus 5T on Android 10.

Klievan, 2021-02-02:
```
I/flutter (10367): REACTIVE_BLE: Start subscribing to notifications for QualifiedCharacteristic(characteristicId: f3641404-00b0-4240-ba50-05ca45bf8abc, serviceId: f3641400-00b0-4240-ba50-05ca45bf8abc, deviceId: 00:0B:57:BE:7E:32)
D/BluetoothGatt(10367): setCharacteristicNotification() - uuid: f3641404-00b0-4240-ba50-05ca45bf8abc enable: true

```
Nothing happens even though 2 notifications have already been sent. If I send more shortly after they do arrive on the app as below:
`I/flutter (10367): REACTIVE_BLE: Received CharacteristicValue(characteristic: QualifiedCharacteristic(characteristicId: f3641404-00b0-4240-ba50-05ca45bf8abc, serviceId: f3641400-00b0-4240-ba50-05ca45bf8abc, deviceId: D9:C3:92:1B:FC:C1), result: CharacteristicValue)`

soyangel, 2021-02-02:
Ok, I think the problem is on `ProtobufMessageConverter.kt`:

```kotlin
 fun createUuidFromParcelUuid(uuid: UUID): pb.Uuid {
        System.out.println("createUuidFromParcelUuid: uuid=" + uuid)
        val convertedUuid = uuidConverter.byteArrayFromUuid(uuid)
        val byteArray = byteArrayOf(convertedUuid[positionMostSignificantBit],
                convertedUuid[positionLeastSignificantBit])
        System.out.println("createUuidFromParcelUuid: byteArray.size=" + byteArray.size)
        return pb.Uuid.newBuilder().setData(ByteString.copyFrom(byteArray)).build()
}
```
This prints:

```
System.out: createUuidFromParcelUuid: uuid=02dac273-80b3-475c-83fc-56540d498c68
System.out: createUuidFromParcelUuid: byteArray.size=2
```
Seems like this method is converting a 128bit UUID to byteArray of 2 values.

Klievan, 2021-02-03:
> Seems like this method is converting a 128bit UUID to byteArray of 2 values.

Correct, by only taking byte 3 and 4. Replacing the last line with below hacks all UUID's to be 128-bit. Verified and tested
`return pb.Uuid.newBuilder().setData(ByteString.copyFrom(byteArray)).build()`
A full-fix would be checking against the BLE base-UUID and if it matches, only return the short-UUID. Else return the whole UUID.

`00000000-0000-1000-8000-00805F9B34FB`

werediver, 2021-02-03:
@soyangel Thanks for tracing this down.

@Klievan The solution you presented looks good 👍 I doubt we should mess with the IDs in _any_ way, though. I wouldn't discard the base UUID part, if it's present. One argument against is people may as well expect the full UUIDs instead of short ones.

@remonh87 It seems the root cause is discovered here. You asked me about the corresponding iOS implementation. Here you are: [ios/Classes/Plugin/PluginController.swift#L270](https://github.com/PhilipsHue/flutter_reactive_ble/blob/35cf9908a465e44b5c44f2d58a8f28ede7f78d39/ios/Classes/Plugin/PluginController.swift#L270)

Papajik, 2021-02-03:
Returning all UUIDs as 128 bit would be much better since you can just extend 16 or 32 bit UUID to 128 bit easily in your own code rather than trying to edit peripheral (which you often can not). 
Only problem are short UUIDs. I can imagine someone is using short UUID to subscribe/write/read characteristic, but wants to use long UUID to check if UUID is among returned values of service discovery. 

Would this factory function in **uuid.dart** help? 

```
/// Returns 128-bit version of given UUID 
 factory Uuid.extendFromUuid(Uuid short){
    var uuidBase = [0, 0, 0, 0, 0, 0, 16, 0, 128, 0, 0, 128, 95, 155, 52, 251];
    if (short.data.length == 2){
      uuidBase.replaceRange(2, 4, short.data);
      return Uuid(uuidBase);
    }
    
    if (short.data.length == 4) {
      uuidBase.replaceRange(0, 4, short.data);
      return Uuid(uuidBase);
    }

    return short;
  }
```
Tested (since its paramater is also Uuid there are only 3 possible versions of given UUID)
```
    var short32 = Uuid.parse('ABCDABCD');
    var short16 = Uuid.parse('ABCD');
    var long = Uuid.parse('50A00101-4E12-11EB-AE93-0242AC130002');
    var extended32 = Uuid.extendFromUuid(short32);
    var extended16 = Uuid.extendFromUuid(short16);
    var extended128 = Uuid.extendFromUuid(long);
    print('Short16: \t' + short16.toString() + '\t -> ' + extended16.toString());
    print('Short32: \t' + short32.toString() + ' -> ' + extended32.toString());
    print('Long: ' + long.toString() + ' -> ' + extended128.toString());

   ///Output
   I/flutter (16284): Short16: 	abcd	 -> 0000abcd-0000-1000-8000-00805f9b34fb
   I/flutter (16284): Short32: 	abcdabcd -> abcdabcd-0000-1000-8000-00805f9b34fb
   I/flutter (16284): Long: 	50a00101-4e12-11eb-ae93-0242ac130002 -> 50a00101-4e12-11eb-ae93-0242ac130002
```
I can imagine to use this as: 
```
var services = await _flutterReactiveBle.discoverServices(deviceId);
var discoveredService = services.firstWhere((service) =>
    service.serviceId == Uuid.extendFromUuid(_settingsService.SERVICE_UUID),
        orElse: () => null);
```
This way you can just convert Uuid when necessary and work with short UUID in other parts of app.

werediver, 2021-02-03:
@Papajik I prefer to keep this library as transparent as possible, unless it doesn't allow a uniform interface across the supported platforms.

If you have a strong opinion about UUID normalization, you can implement it on the application side.

Papajik, 2021-02-03:
I was not trying to suggest platform specific code. I  just initially thought you can not use 128-bit UUID for services/characteristics which are advertised as 16-bit (I got some errors when I tried it at first but I found out it was mistake in my code) and I tried to present a solution in which you don't have to use two UUID for same characteristic. One long for write/read/subscribe and second for comparing result of discoveryService. But you are already coverting everything to long UUID on [platform specific code here](https://github.com/PhilipsHue/flutter_reactive_ble/blob/35cf9908a465e44b5c44f2d58a8f28ede7f78d39/android/src/main/kotlin/com/signify/hue/flutterreactiveble/converters/UuidConverter.kt#L15) because of Android needs.  So you are right my previous comment doesn't make much sense when `Uuid.parse('2A00')` and `Uuid.parse('00002A00-0000-1000-8000-00805F9B34FB')` have in the end same behaviour and you can use longer version for both comparison and function calls.

werediver, 2021-02-03:
@remonh87 I don't exactly like how the that Android UUID conversion looks ([converters/UuidConverter.kt#L15](https://github.com/PhilipsHue/flutter_reactive_ble/blob/35cf9908a465e44b5c44f2d58a8f28ede7f78d39/android/src/main/kotlin/com/signify/hue/flutterreactiveble/converters/UuidConverter.kt#L15)). It's no really "transparent", it indeed makes everything 128-bit, as mentioned by @Papajik. We should _just copy the bytes_ (see the any UUID conversion on iOS).

I believe, the solution mentioned by @Klievan [abbove](https://github.com/PhilipsHue/flutter_reactive_ble/issues/209#issuecomment-772464612) is a good approach.

Shall we get the UUID conversion on Android in line with how it's done on iOS, @remonh87?

remonh87, 2021-02-04:
Thanks all for the help on the issue. I integrated a fix in master. Can someone check if this solves the issue before shipping a new version to pub.dev?

closed 2021-02-04 by remonh87

soyangel, 2021-02-04:
In my app I'm only using long UUIDs and it's working perfectly:

```
DiscoveredService(serviceId: 00dac273-80b3-475c-83fc-56540d498c68, characteristicIds: [01dac273-80b3-475c-83fc-56540d498c68, 02dac273-80b3-475c-83fc-56540d498c68], includedServices: [])
```

Thanks guys, you rule!  👏👏👏

Klievan, 2021-02-04:
Worked for me too. Thank you so much for addressing this so quickly guys. 👏

safield, 2021-02-04:
I am experiencing this issue as well. Is a new release imminent?

Papajik, 2021-02-04:
Thank you guys for solving it so quickly

PR #217 (2021-02-04): fix bug in uuid conversion on android
Fix #209

I really do not know why I made it before like this it drives me nuts this part of the code. Really need some time to refactor things there