PhilipsHue/flutter_reactive_ble #211: First quick notifications seem to be ignored

Klievan, 2021-02-02:
**Describe the bug**
I have a BLE device that instantly starts sending notifications once they are turned on. It seems that the first 1-2 notifications seem to be not sent to the stream I am listening to.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to a BLE device that instantly sends notifications when written to the CCCD to enable notifications
2. Subscribe to the characteristic and print out all incoming notifications
3. Observe that the first data does not seem to be received 

**Expected behavior**
I expect the first notifications to be also received.

- [ ] When using nRFConnect or flutter_blue I am able to receive all notifications.

**Smartphone / tablet**
 - Device: OnePlus 5T
 - OS:  Android 10
 - Package version: ^2.7.2

**Peripheral device**
- Vendor, model: Both the Silabs BGM121 and nRF52832 appear to have the same behavior 
- Does it run a custom firmware / software: yes

remonh87, 2021-02-02:
This is a very specific usecase and I really need your help in triaging this. Can you tryout the same usecase but then set the loglevel to verbose?  you can use the setter `logLevel` on the `FlutterReactiveBle` client for this

Klievan, 2021-02-02:
```
I/flutter (10367): REACTIVE_BLE: Start subscribing to notifications for QualifiedCharacteristic(characteristicId: f3641404-00b0-4240-ba50-05ca45bf8abc, serviceId: f3641400-00b0-4240-ba50-05ca45bf8abc, deviceId: 00:0B:57:BE:7E:32)
D/BluetoothGatt(10367): setCharacteristicNotification() - uuid: f3641404-00b0-4240-ba50-05ca45bf8abc enable: true

```
This appears but no notifications are received (even though 2 already have been sent). The ones that arrive about 200ms later do get registered though. The use case is a file being streamed through notifications once they are enabled, but the first notifications seem to get accepted by the phone but do not get passed onto the app.

remonh87, 2021-02-03:
hmm I had hoped to see more logging then this 🤔 . If you want I can creste a triage branch with verbose logging but I may be able to do this the coming days. Problem is that I do not have the exact same peripheral which makes it difficult to check.

Klievan, 2021-02-03:
I can be actively debugging the issue with you if needed, or if you happen to have an nRF52 devkit of some sorts I could send you a firmware example. I've further tested this on an iOS device and I get the following assert:
```
Assertion failed: file flutter_reactive_ble/PluginController.swift, line 103
2021-02-03 10:31:04.194728+0100 ProxLogs[2891:998648] Assertion failed: file flutter_reactive_ble/PluginController.swift, line 103
```

remonh87, 2021-02-05:
would be great if you can help me out I do not have an nrf52 devkit at the moment so triaging this one would be a bit difficult

werediver, 2021-02-09:
@Klievan I suppose you have control over the peripheral's behavior. _As a workaround_, I'd suggest to try making the peripheral wait a bit before starting streaming data to a new subscription. Think of 50–300 ms.

This is in no way a structural fix, but may allow you continue development for the time being. If it works for you, please let us know. That may be useful in analyzing the issue.

May be related to #137.

Klievan, 2021-04-16:
Waiting 100ms works. Also, I can confirm this is related to #137. On iOS, instead of ignoring the notification the app crashes with an assert. 
Assertion failed: file flutter_reactive_ble/PluginController.swift, line 103
`            onCharacteristicValueUpdate: papply(weak: self) { context, central, characteristic, value, error in
                guard let sink = context.characteristicValueUpdateSink
                else { assert(false); return }
`

closed 2021-08-10 by remonh87

Klievan, 2023-01-22:
Just FYI. This issue is still present, not sure exactly why this was marked as completed. I'm currently trying to figure out where in the plugin this is happening. Not sure if this should be re-opened or if I should create a new issue.