PhilipsHue/flutter_reactive_ble #212: Just works bonding

HelloBytes, 2021-02-02:
I connect to a device that has implemented Just Works pairing. 
On iOS there appears a dialog to pair with the device but on Android this is not the case. 
You can connect with Android but there are no services available. When you connect manually in the bluetooth settings of the Android device you will see the services appear.

What is the problem with Just Works pairing?

werediver, 2021-02-02:
Should Just Work.

@remonh87 Any thoughts?

remonh87, 2021-02-02:
@HelloBytes  Are you testing with a samsung device on Android? I ran into a very interesting blogpost about justworks issues on Samsung devices. https://medium.com/@martijn.van.welie/making-android-ble-work-part-4-72a0b85cb442. Maybe this work around can help you. Else I need some more logs to further triage the issue.

HelloBytes, 2021-02-03:
@remonh87 Thank you for your response and great work.

I'm testing on a pixel 2 and a Huawei tablet. 

So the work around, I have to change the code of the plugin to let it work? 

After connecting to the device I subscribe on a characteristic and I write a characteristic get the following exceptions:

Exception: GenericFailure<CharacteristicValueUpdateError>(code: CharacteristicValueUpdateError.unknown, message: "Characteristic not found with UUID

Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "Characteristic not found with UUID 000030c9-etc..")

This is because it's not paired yet. There is no state about bonding is there?

remonh87, 2021-02-03:
@HelloBytes  we do not have a notion of bonding at the moment in our library because iOS does not export this state as far as I know (right @werediver ? ) . I did check the library of polidea and on Android I really need to force ensurebonding on some devices with justworks. 

@werediver  my suggestion would be to add a flag to `connectToDevice` called `androidEnforceBondingAfterConnect`. What do you think?

werediver, 2021-02-03:
@remonh87 Do you remember how (why) the pairing works in Hue Bluetooth app? Because (I hope! 😅) it does.

I see no point in introducing a flag. What a user would do with it? Set it permanently to `true`? What's the point of having the flag then?

I would rather suggest to unify the library behavior as much as reasonably possible across the platforms. In this case it means handling pairing on android in a more automated way. The article you mentioned above ([ref](https://medium.com/@martijn.van.welie/making-android-ble-work-part-4-72a0b85cb442)) may be a good reference.

What do you think?

HelloBytes, 2021-02-03:
I have no experience in Kotlin but I have tested the code below and the bond is indeed created without a dialog. Only problem after connecting is that the services are not avaiable if the bondstate != 12

`private fun connectDevice(rxBleDevice: RxBleDevice, shouldNotTimeout: Boolean): Observable<RxBleConnection>  {

        var bondState: Int = rxBleDevice.getBluetoothDevice().getBondState()

        println("bond state")
        println(bondState)

        if (bondState != 12) {
            println("create bond")
            if (rxBleDevice.getBluetoothDevice().createBond()) {
                return rxBleDevice.establishConnection(shouldNotTimeout)
                        .compose {
                            if (shouldNotTimeout) {
                                it
                            } else {
                                it.timeout(
                                        Observable.timer(connectionTimeout.value, connectionTimeout.unit),
                                        Function<RxBleConnection, Observable<Unit>> {
                                            Observable.never<Unit>()
                                        }
                                )
                            }
                        }
            }
        }
        println("bond already created")
        return rxBleDevice.establishConnection(shouldNotTimeout)
                .compose {
                    if (shouldNotTimeout) {
                        it
                    } else {
                        it.timeout(
                                Observable.timer(connectionTimeout.value, connectionTimeout.unit),
                                Function<RxBleConnection, Observable<Unit>> {
                                    Observable.never<Unit>()
                                }
                        )
                    }
                }

    }`

remonh87, 2021-02-16:
@werediver  @HelloBytes  I did some investigation and the only thing that I can think of is that discoverservices is called when the state is bonding which can trigger an interupt. What I can do (and actually should do for all operations) is throw an exception in case the bonding state is bonding. For the rest I do not see why just works pairing fails, like the name said it should just work :) 

@HelloBytes  can you confirm to me that when discoverservices do not work the bondstate is bonding  (bondstate==11)?

HelloBytes, 2021-02-17:
@remonh87 I checked it and it was == 10 (bond none)

https://developer.android.com/reference/android/bluetooth/BluetoothDevice#BOND_NONE

HelloBytes, 2021-02-19:
@remonh87 @werediver is there code I can test at this moment? I really have problems on android connecting to a device that needs bonding.

remonh87, 2021-02-20:
@HelloBytes the problem is that discoverservices is called when the device is bonding. So I created a fix that will prevent this from happening in the future (you will receive an error). 

What I also would suggest is to make sure you call discover services after a short delay (suggest 500ms but play a bit with it) in order to give the device time to bond. Based on the articles that I read around just works it should work on our library as well without any necessarry additions from our side.

HelloBytes, 2021-02-21:
@HelloBytes thank you. I will test it.

So there is no need to call createBond()?

remonh87, 2021-02-21:
No the peripheral should indicate after connecting it wants to establish a bond and then the Android OS pics this up

remonh87, 2021-03-10:
I did this weekend some playing around with just works pairing and I got the popup and it worked for me like I described. I will close this for now.

closed 2021-03-10 by remonh87