PhilipsHue/flutter_reactive_ble #213: VSCode Android build fails to build (debug and release mode)

plyrek, 2021-02-02:
**Describe the bug**
A clear and concise description of what the bug is.
Trying to build to an Android Simulator or Samsung S10 completes with 2 failures. iOS build fine

**To Reproduce**
Steps to reproduce the behavior:
1. Click run in VSCode when attached to Android simulator or Android device

**Expected behavior**
Builds a debug or release APK and installs on device.

 - Device: Android Samsung S10 or Nexus 6 Simulator running API 30
 - OS: Android 30.0.3
 - Package version: 2.7.3

**Additional context**
**Debug Console Output:**

Launching lib/main.dart on SM G975U in debug mode...
 lib/main.dart

FAILURE: Build completed with 2 failures.
1: Task failed with an exception.
-----------
* Where:
Build file '/Users/development/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_reactive_ble-2.7.3/android/build.gradle' line: 35

* What went wrong:
A problem occurred evaluating project ':flutter_reactive_ble'.
> Failed to apply plugin [id 'de.mannodermaus.android-junit5']
   > android-junit5 plugin requires Gradle 6.1.1 or later

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
A problem occurred configuring project ':flutter_reactive_ble'.
> Failed to notify project evaluation listener.
   > compileSdkVersion is not specified.
   > Extension not initialized yet, couldn't access compileSdkVersion.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* Where:
Build file '/Users/development/flutter/.pub-cache/hosted/pub.dartlang.org/flutter_reactive_ble-2.7.3/android/build.gradle' line: 35

* What went wrong:
A problem occurred evaluating project ':flutter_reactive_ble'.
> Failed to apply plugin [id 'de.mannodermaus.android-junit5']
   > android-junit5 plugin requires Gradle 6.1.1 or later

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
A problem occurred configuring project ':flutter_reactive_ble'.
> Failed to notify project evaluation listener.
   > compileSdkVersion is not specified.
   > Extension not initialized yet, couldn't access compileSdkVersion.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org

BUILD FAILED in 590ms
Exception: Gradle task assembleDebug failed with exit code 1

**Flutter Doctor -v Output**

[✓] Flutter (Channel stable, 1.22.6, on macOS 11.1 20C69 darwin-x64, locale en-US)
 
[✓] Android toolchain - develop for Android devices (Android SDK version 30.0.3)
[✓] Xcode - develop for iOS and macOS (Xcode 12.4)
[!] Android Studio (version 4.1)
    ✗ Flutter plugin not installed; this adds Flutter specific functionality.
    ✗ Dart plugin not installed; this adds Dart specific functionality.
[✓] VS Code (version 1.52.1)
[✓] Connected device (2 available)
    ! Error: Paul’s iPhone is busy: Copying cache files from device. Xcode will continue when Paul’s iPhone is finished. (code -10)

! Doctor found issues in 1 category.
immergenmac2@Pauls-MacBook-Pro mobile-flutter-afx2020 % flutter doctor -v
[✓] Flutter (Channel stable, 1.22.6, on macOS 11.1 20C69 darwin-x64, locale en-US)
    • Flutter version 1.22.6 at /Users/development/flutter
    • Framework revision 9b2d32b605 (11 days ago), 2021-01-22 14:36:39 -0800
    • Engine revision 2f0af37152
    • Dart version 2.10.5

[✓] Android toolchain - develop for Android devices (Android SDK version 30.0.3)
    • Android SDK at /Users/immergenmac2/Library/Android/sdk
    • Platform android-30, build-tools 30.0.3
    • Java binary at: /Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b3-6915495)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 12.4)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • Xcode 12.4, Build version 12D4e
    • CocoaPods version 1.10.0

[!] Android Studio (version 4.1)
    • Android Studio at /Applications/Android Studio.app/Contents
    ✗ Flutter plugin not installed; this adds Flutter specific functionality. _(FYI this is know VScode issue with Flutter doctor has always been this way even when build was working)._
    ✗ Dart plugin not installed; this adds Dart specific functionality. _(FYI this is know VScode issue with Flutter doctor has always been this way even when build was working)._
    • Java version OpenJDK Runtime Environment (build 1.8.0_242-release-1644-b3-6915495)

[✓] VS Code (version 1.52.1)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.19.0

HelloBytes, 2021-02-03:
In android studio as well

remonh87, 2021-02-03:
@HelloBytes  you need from now on gradle wrapper 6.1.1 or later. Check your gradle-wrapper properties file and upgrade this accordingly. Also your build.gradle needs to have a proper compile sdk

plyrek, 2021-02-03:
@remonh87 Can confirm that upgrading to the latest gradle wrapper version fixed the compile issue.

gradle-wrapper.properties file line 6: distributionUrl=https\://services.gradle.org/distributions/gradle-6.8.1-bin.zip

I did not have to change the build.gradle compile sdk however. It was already set correctly. Must have been a byproduct of the gradle issue.

Thank you.

remonh87, 2021-02-03:
cool happy it works. It is a bit annyoing that google is not consistent in upgrading gradle for their total productline because some product require a higher version of gradle 6.5 while flutter has by default a lower version.

closed 2021-02-03 by remonh87