PhilipsHue/flutter_reactive_ble #214: Android: crash with 2.7.3

il-picchio, 2021-02-03:
**Describe the bug**
Only with Android.
The app crashes using the plugin version 2.7.3, with 2.7.2 it works flawlessly (great job, the best right now).
The crash occurs only when the app is built in release mode (--release).

**To Reproduce**
Steps to reproduce the behavior:
1. Init reactive client
2. Start to scan

**Expected behavior**
The app doesn't crash as previous plugin version :D

**Smartphone / tablet**
 - Device: Nokia 4.2 / Samsung A51
 - OS: Android 10 / Android 10
 - Package version: 2.7.3

**Peripheral device**
- Vendor, model: CUSTOM
- Does it run a custom firmware / software: yes

**Additional context**
StackTrace:

Fatal Exception: d.b.z.a: 2 exceptions occurred. 
       at io.reactivex.internal.observers.LambdaObserver.a(LambdaObserver.java:32)
       at io.reactivex.internal.observers.LambdaObserver.a(LambdaObserver.java:25)
       at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.drainNormal(ObservableObserveOn.java:47)
       at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.run(ObservableObserveOn.java:8)
       at io.reactivex.android.schedulers.HandlerScheduler$ScheduledRunnable.run(HandlerScheduler.java:2)
       at android.os.Handler.handleCallback(Handler.java:883)
       at android.os.Handler.dispatchMessage(Handler.java:100)
       at android.os.Looper.loop(Looper.java:214)
       at android.app.ActivityThread.main(ActivityThread.java:7410)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)

----------------------

Caused by d.b.z.a$a: Chain of Causes for CompositeException In Order Received =>
       at com.google.firebase.crashlytics.internal.stacktrace.TrimmedThrowableData.<init>(TrimmedThrowableData.java:29)
       at com.google.firebase.crashlytics.internal.common.CrashlyticsReportDataCapture.a(CrashlyticsReportDataCapture.java:18)
       at com.google.firebase.crashlytics.internal.common.SessionReportingCoordinator.a(SessionReportingCoordinator.java:19)
       at com.google.firebase.crashlytics.internal.common.SessionReportingCoordinator.a(SessionReportingCoordinator.java:32)
       at com.google.firebase.crashlytics.internal.common.CrashlyticsController$6.call(CrashlyticsController.java:53)
       at com.google.firebase.crashlytics.internal.common.CrashlyticsController$6.call(CrashlyticsController.java)
       at com.google.firebase.crashlytics.internal.common.CrashlyticsBackgroundWorker$3.then(CrashlyticsBackgroundWorker.java:2)
       at com.google.android.gms.tasks.zzf.run(zzf.java:8)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
       at com.google.firebase.crashlytics.internal.common.ExecutorUtils$1$1.onRun(ExecutorUtils.java:2)
       at com.google.firebase.crashlytics.internal.common.BackgroundPriorityRunnable.run(BackgroundPriorityRunnable.java:5)
       at java.lang.Thread.run(Thread.java:919)

----------------------

Caused by java.lang.RuntimeException: Field status_ for c.f.a.a.c not found. Known fields are [private static final c.f.a.a.c c.f.a.a.c.e, private static volatile c.b.b.a1 c.f.a.a.c.f]
       at com.google.protobuf.MessageSchema.a(MessageSchema.java:72)
       at com.google.protobuf.MessageSchema.a(MessageSchema.java:716)
       at com.google.protobuf.MessageSchema.a(MessageSchema.java:12)
       at com.google.protobuf.ManifestSchemaFactory.a(ManifestSchemaFactory.java:60)
       at com.google.protobuf.ManifestSchemaFactory.a(ManifestSchemaFactory.java:49)
       at com.google.protobuf.Protobuf.a(Protobuf.java:17)
       at com.google.protobuf.Protobuf.a(Protobuf.java:4)
       at com.google.protobuf.GeneratedMessageLite.makeImmutable(GeneratedMessageLite.java:4)
       at com.google.protobuf.GeneratedMessageLite$Builder.k(GeneratedMessageLite.java:9)
       at com.google.protobuf.GeneratedMessageLite$Builder.build(GeneratedMessageLite.java)
       at com.signify.hue.flutterreactiveble.channelhandlers.BleStatusHandler$listenToBleStatus$2.accept(BleStatusHandler.java:11)
[... 128 lines cut ...]
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1152)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
       at java.lang.Thread.run(Thread.java:919)

-----------------------

Plugins used:

  flutter_secure_storage: ^3.3.3
  flutter_reactive_ble: ^2.7.3
  permission_handler: ^5.0.1+1
  provider: ^4.3.2+2
  carousel_slider: ^2.2.1
  percent_indicator: ^2.1.7+4
  fl_chart: ^0.12.0
  http: ^0.12.2
  async: ^2.4.2
  firebase_core: ^0.7.0
  firebase_crashlytics: ^0.4.0+1
  sqflite: ^1.3.2+1
  shared_preferences: ^0.5.12+4
  path: ^1.7.0
  crypto: ^2.1.5
  url_launcher: ^5.7.10
  package_info: ^0.4.3+4

remonh87, 2021-02-03:
Hmm that sounds like an proguard issue can you try to add `-keep class com.google.protobuf.** { *; }`  to your pro-guard-rules file? It is a bit probably linked to the fact firebase uses some outdated version from one or the other. I will do some triage in the coming days as well.

remonh87, 2021-02-04:
@il-picchio  I did some triage on the exampel app and added the same depedencies to the pubspec.yaml as you did but for me the app runs normal. I would check your proguard rules and update gradle to version 6.5.1 this should hopefully fix the issue.

il-picchio, 2021-02-04:
Hello, 
I am not using proguard in this project yet so I can't upgrade any rule unfortunatly

remonh87, 2021-02-05:
I think proguard is on by default in release (if I am not mistaken) and this issue is related to that one. Also can you check your android tools version and gradle wrapper version ? For this library I recommend gradle-wrapper 6.5.1 and android tools version 4.1.2

closed 2021-02-13 by remonh87

il-picchio, 2021-02-16:
Hello,

sorry for the delayed reply.
I confirm that the problem is not proguard (default is R8 though, proguard should be deprecated or whatever) which I am not using.
Finally I figured out what's going on. It was a problem in my gradle settings.

So thank you for your help, you have done a really good job with this plugin