PhilipsHue/flutter_reactive_ble #215: No update is broadcast after timeout.

safield, 2021-02-03:
After using the method connectToAdvertisingDevice, and providing a timeout, once the timeout expires, no update is broadcast to the subscription to the stream.

The expected behaviour would be that an update would be broadcast to the stream indicating a timeout.

Observe the following code.

```
      var stream = _ble.connectToAdvertisingDevice(
        id: device.id,
        withServices: [ ],
        prescanDuration: const Duration(seconds: 4),
        connectionTimeout: const Duration(seconds:  4),
      );
      
      _deviceSubscription = stream.listen((ConnectionStateUpdate update) {
         print("this never prints");
      }); // only use this subscription to disconnect
```

Either ConnectionStateUpdate should arrive with some sort of value for failure, or connectToAdvertisingDevice should take a callback onFailure parameter.

werediver, 2021-02-04:
A timeout usually leads to an exception in the stream. Do mean no exceptions are being thrown in case of a timeout in `connectToAdvertisingDevice(⋯)`?

Is this common to both iOS and Android? (if possible, please check)

safield, 2021-02-04:
Not possible to check iOS at the moment.

I have updated my initial post to include the fact that my call is wrapped in a try/catch. No exception is caught though. I am a little confused though (potentially due to my limited knowledge on streams and isolates), but how do I catch en exception, when the try/catch will exit long before the timeout actually occurs?

Either way....there is a log write about an exception...but no actual exception.

`E/flutter (23659): [ERROR:flutter/lib/ui/ui_dart_state.cc(177)] Unhandled Exception: TimeoutException after 0:00:03.000000: Future not completed.`

I am running this under the vscode debugger. I have the settings 'break on all exceptions' on, in addition, at the top level, my app is running in a zone, which is not catching any exception either.

werediver, 2021-02-04:
Please, check the documentation on [Stream.listen(⋯)](https://api.flutter.dev/flutter/dart-async/Stream/listen.html).

safield, 2021-02-04:
Thanks! That was the issue. I have updated the comments to hopefully make it harder for future muggles like me to make the same mistake / oversight.

Cheers!

closed 2021-02-04 by safield