PhilipsHue/flutter_reactive_ble #216: decoding of device manufacturerData returns only SINGLE manfacturerData block when there can be MULTIPLE

timmaffett, 2021-02-04:
The device manufacturerData data is a single Uint8List.   The manufacturerData for a device can be a *LIST* of data.

The way the flutter_blue package handles this correctly by having manufacturerData be Map<int, List<int>>

flutter_reactive_ble treats the *single* returned data block as a Uint8List, where the first two bytes of the data would be the manufacturerId (not parsed out).    
When there are multiple manufacturerData blocks ONLY 1 is captured and the rest are LOST.

Bluetooth devices like the Govee temperature/humidity sensors place the temperature and humidity data with manufacturerData blocks within the advertisements.  The flutter_reactive_ble package cannot handle devices that have multiple manufacturerData blocks.  (It does correctly handle the multiple serviceData blocks, it just needs to be fixed to handle the mutliple manufacturerData blocks.

There are two ways to fix this bug.
1) In parsing Advertisement catch each manafacturerData block and instead return a List<Uint8List> with each block parsed no further (like how it is handled now, but correctly catching ALL of the manufacturerData blocks)
(In ManufacturerDataConverter.kt the companyId is actually RECOMBINED into the first 2 bytes of the rawData manufacturerData block)

or 

2) Parse each of the manufactureData blocks into the manufacturerId (first 2 bytes (short) of each block) and return a Map of the manufacturerId's (called companyId ManufacturerDataConverter.kt) to Uint8List   ( Map< int, Uint8List > )  (this seems more useful way to do this.  The ServiceDataEntry's are parsed into UUID and the data playload,  a ManufacturerDataEntry could be created the same way.)

werediver, 2021-02-04:
This is common for both iOS and Android, correct?

xvrh, 2021-04-15:
If I remember correctly, we decided to only add a single manufacturerData entry because iOS only support one via [CBAdvertisementDataManufacturerDataKey](https://developer.apple.com/documentation/corebluetooth/cbadvertisementdatamanufacturerdatakey) (but I can be proved wrong)

Maybe we can modify the API to be like:

```
class DiscoveredDevice {
  List<ManufacturerDataEntry> get manufacturerData;
}
class ManufacturerDataEntry {
  int get companyId;
  Uint8List get data;
}
```

FXschwartz, 2022-11-21:
Hey all, is there any possibility this is on the roadmap in the foreseeable future? I've just got a project, and the ble devices have multiple manufacturer data blocks.