PhilipsHue/flutter_reactive_ble #219: It would be nice if subscribeToCharacteristic threw an error on invalid characteristic data.

safield, 2021-02-04:
When using the _subscribeToCharacteristic_, no error is thrown when invalid data is used.

_subscribeToCharacteristic_ take a QualifiedCharacteristic as its sole parameter. I can pass a QualifiedCharacteristic with an invalid serviceId, characteristicId, or deviceId. In this scenario the library just silently fails. None of the callbacks provided to the stream are ever called.

My concrete example, is that I have a specific service that contains a specific characteristic. If the service exists, but the characteristic doesn't, I consider this an error case that I would explicitly like to handle. In the current implementation, I must use _discoverServices_ for the sole reason to validate that the expected services and characteristics exist.

Ideally, _subscribeToCharacteristic_ would be able to handle that for me, and reduce work on my end.

werediver, 2021-03-29:
Is this behavior specific to a platform? It doesn't seem to be an improvement, but rather a bug, @remonh87.

remonh87, 2021-03-29:
yes it is a bug we should investigate this.  Do not know why I labeled this an enhancement