PhilipsHue/flutter_reactive_ble #221: Calling scanForDevices() twice crashes

lukepighetti, 2021-02-06:
If you call scanForDevices() twice it crashes the app.

Only tested on iOS.

```
Assertion failed: file flutter_reactive_ble/PluginController.swift, line 150
Assertion failed: file flutter_reactive_ble/PluginController.swift, line 150
* thread #1, queue = 'com.apple.main-thread', stop reason = Assertion failed
    frame #0: 0x00000001a863aff8 libswiftCore.dylib`_swift_runtime_on_report
libswiftCore.dylib`_swift_runtime_on_report:
->  0x1a863aff8 <+0>: ret    
libswiftCore.dylib`_swift_reportToDebugger:
    0x1a863affc <+0>: b      0x1a863aff8               ; _swift_runtime_on_report
libswiftCore.dylib`_swift_shouldReportFatalErrorsToDebugger:
    0x1a863b000 <+0>: adrp   x8, 352927
    0x1a863b004 <+4>: ldrb   w0, [x8, #0x998]
Target 0: (Runner) stopped.
```

werediver, 2021-02-09:
Please, make sure you don't attempt to run more than one scan at a time. It is fine to start a scan, end it, then start another one.

Let us know if that solves your issue.

lukepighetti, 2021-02-10:
How do you end a scan?

How do you check if a scan is already in progress?

werediver, 2021-02-10:
You end a scan the same way as you end a connection: you unsubscribe from the stream returned by a method call.

A scan is in progress, if you have started it, but have not ended it.

lukepighetti, 2021-02-10:
Great, thanks!

closed 2021-02-10 by lukepighetti