PhilipsHue/flutter_reactive_ble #222: Improve description, examples, and code comments related to BleStatus.

scottsidekick, 2021-02-09:
The BleStatus enum and the status subscription are important to getting a fully function app working using bluetooth.

But the current example and descriptions of the status are not quite thorough enough to help a newbie figure how to get things working properly.

For example, I am particularly confused around the 'unauthorized' value, and what the proper course of action should be when this value is received. Does this mean that I need to prompt the user for some sort of platform specific permissions? Or does this mean that I did not add proper permissions in my apps platform specific config files (eg AndroidManifest.xml)?

```
    poweredOff - tell user to turn on bluetooth?
    ready - good to go.
    unauthorized - ??
    unknown - put ui in loading state until this had changed.
    unsupported - tell user app won't work on their device.
```

In the front page readme example, it might be nice to add a switch statement to the BleStatus subscription example, with at least comments explaining potential course of actions.

Thanks!! :)

werediver, 2021-02-09:
On iOS `BleStatus` translates to/from [CBManagerState](https://developer.apple.com/documentation/corebluetooth/cbmanagerstate) type. Please, check the source code to find to which type it translates on Android.

"Unauthorized" on iOS means the first time the app tried to use BLE and the system asked user whether to allow that app to use it, user denied the app to access BLE. On iOS the app cannot change this. The app can only instruct the user to go to the system settings and grant the permission manually.

remonh87, 2021-02-09:
Small addition to the above comment.  We try to keep the readme documentation brief to cover the basics and we put a lot of effort in the documentation in the [sourcecode](https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/lib/src/model/ble_status.dart) which also ends up in [dartdoc](https://pub.dev/documentation/flutter_reactive_ble/latest/flutter_reactive_ble/BleStatus-class.html). If you still are missing something let us know.

remonh87, 2021-06-28:
I checked the source code and the description is quite self explaining will leave it for now

closed 2021-06-28 by remonh87