PhilipsHue/flutter_reactive_ble #223: Calling cancel on ConnectionStateUpdate stream results in PlatformException

scottsidekick, 2021-02-11:
I am connecting to a device as follows....

```
        var stream = _ble.connectToAdvertisingDevice(
          id: device.id,
          withServices: [ ],
          prescanDuration: const Duration(seconds: 20),
          connectionTimeout: const Duration(seconds:  20),
        );
        
        _deviceSubscription = stream.listen(
          _connectionStateUpdateListener,
          cancelOnError: true
        ); 
```

Then later on I disconnect as follows...

```
    if (_scanSubscription != null)
      await _scanSubscription.cancel();
    if (_deviceSubscription != null)
      await _deviceSubscription.cancel();
    if (_readCharSubscription != null)
      await _readCharSubscription.cancel();
```

After the deviceSubscription is cancelled, a platform exception is thrown.

```
Exception has occurred.
PlatformException (PlatformException(flutter_reactive_ble.Central.(unknown context at $1093a3cf0).Failure:1, The operation couldn’t be completed. (flutter_reactive_ble.Central.(unknown context at $1093a3cf0).Failure error 1.), {}, null))
```

werediver, 2021-02-11:
You are mentioning some `_readCharSubscription`. I suppose it's a subscription to a stream returned by `subscribeToCharacteristic(·)` (because just reading does not involve streams). If this is correct, you are getting an exception in this stream, because it is not possible to listen to characteristic's updates when the connection is terminated (intentionally or not, doesn't matter).

When you are working with IO you should be prepared to handle failures.

scottsidekick, 2021-02-11:
So is there a suggested way to disconnect? I have changed the order to call cancel on _readCharSubscription first, and the exception is still thrown.

Should I just never ever call cancel on a subscription to a characteristic?

werediver, 2021-02-11:
If you make sure that you close all other subscriptions related to a particular device before you disconnect (which means `await` on [cancel()](https://api.dart.dev/stable/2.10.5/dart-async/StreamSubscription/cancel.html) calls), you shouldn't get exceptions due to loss of connection (introduce a synthetic delay of about a couple hundreds milliseconds after closing everything and before disconnecting, if `awaiting` is not enough [though, that would be a workaround and a bug in the library, that is worth reporting]).

But note, that the connection doesn't only get terminated, because you close it. It gets terminated, because a peripheral goes out of range or powers off or due to heavy RF interference, etc. This is why you should be prepared for failures when performing IO operations. There _will_ be failures.

If there is a bug, please provide enough details so that we can understand and investigate it. Do not disregard the issue template, because it may answer some of the questions we would anyway ask you. Right now I do not see anything criminal in the described behavior (as far as I can tell from the details provided).

At the very least, you should clarify, in which exactly stream the exception occurs, because that is very important.

scottsidekick, 2021-02-11:
If I add a delay it works...

```
    if (_readCharSubscription != null)
      await _readCharSubscription.cancel();
    await Future.delayed(Duration(seconds: 5));
    if (_deviceSubscription != null)
      await _deviceSubscription.cancel();
```

If I remove a delay it throws an exception....

```
    if (_readCharSubscription != null)
      await _readCharSubscription.cancel();
    if (_deviceSubscription != null)
      await _deviceSubscription.cancel();
```

Based on your previous comments....this is a bug? Do you want this submitted as a fresh bug using the template?

werediver, 2021-02-11:
Thank you for doing the additional tests. I'll reopen this ticket, but it would help if you add this detail about the delay into the original post.

This seems to be related to [a discussion in #157](https://github.com/PhilipsHue/flutter_reactive_ble/issues/157#issuecomment-770109198). As it pops up the second time, it may be worth reworking / fixing the disconnect operation to complete when the disconnecting has been actually completed.

scottsidekick, 2021-02-11:
"At the very least, you should clarify, in which exactly stream the exception occurs, because that is very important".

I am actually having trouble doing this. I wrapped both cancels in individual try/catch AND in individual runZoned....but the exception is not actually being caught :/

werediver, 2021-02-11:
Please, check the documentation on [Stream.listen(⋯)](https://api.flutter.dev/flutter/dart-async/Stream/listen.html).

You cannot use try-catch around a call to `Stream.listen(⋯)` to catch errors in a stream, because the call completes before events start to come into `onData:` and `onError:` handlers.

scottsidekick, 2021-02-17:
I can confirm that the error is thrown from the stream with subscription _readCharSubscription.