PhilipsHue/flutter_reactive_ble #227: Get the list of bonded/paired devices

Wassim24, 2021-02-15:
**Is your feature request related to a problem? (please describe)**
A clear and concise description of what the problem is. Ex. I'm always frustrated when i can't get the list of bonded devices from android or iOS.

**Describe the solution you'd like**
Using the flutter reactive ble instance to get the list of the bonded devices, for examples: `_ble.getBondedDevices()`

**Describe alternatives you've considered**
An alternative we considered is creating a plugin that uses the method channels to provide this list, the inconvenient of this solution, is that it is not "well integrated" with your library.

**Additional context**
In our use case, the user has a list of sensors, but in certain circumstances some of them may not have been bonded. Having access to the list of bonded devices we can enhance the user experience, by telling the user which sensor is bonded or not.

Thank you in advance for your help, and this fantastic library
Cheers

werediver, 2021-02-15:
Querying the system for the bonded BLE peripherals is not possible on iOS and is probably not possible on Android (@remonh87 please, comment).

An application should store the list of devices it discovered and connected to. Even when a peripheral is already bonded with a smartphone, you can still discover it and connect the usual way.

remonh87, 2021-02-15:
@werediver  @Wassim24 On Android it is indeed possible to get a list of bonded devices. From Android I also partly understand the use case however can you explain the problem bonded devices would solve in your case? For example in our Hue app we also have lights that are bonded or not bonded but we saw no need to implement this functionality.

werediver, 2021-02-15:
Implementing this functionality for Android only would also make the library behavior pretty asymmetrical between platforms.

Wassim24, 2021-02-15:
Thank you very much @remonh87 & @werediver for you quick responses :)

The use case, is that when accessing a certain screen, the application starts searching a known list of devices. When it finds a device it connects to it, starts reading some characteristics periodically, and displays the device's data.
The problem is: if a user forgets one of our devices from the android Bluetooth menu, then android will display the pairing popup whenever the application tries to connect to this forgotten device, which can be annoying if we have a lot of devices.

We thought that we can avoid this kind of behavior, by avoiding to connect to unbonded devices (if we had the bonded devices list), and indicate to the user which devices has been unbonded.

Also, as our user's devices are stored in the cloud, if he changes his phone, the application does not know if the devices are bonded or not.

Thank you in advance :)

remonh87, 2021-02-15:
I understand your use case now thanks for the explanation. Still I am not sure if you want to deviate in behavior between Android and iOS in this case. Since iOS does not handle this use case you still end up with a suboptimal solution for iOS users. That's why I am also doubting to implement this usecase in our library. One of the strengths of this library is that it behaves consistent on iOS and Android and by implementing bonding functionality we would deviate from this principle. @werediver  what do you think?

werediver, 2021-02-15:
On iOS the IDs of BLE peripherals are always _randomized_ (IDs of bonded devices are remembered, non-bonded devices may get new IDs on every discovery). On Android, as far as I know, the IDs of BLE peripherals _may_ be randomized (even if you don't observe it on your development device with a particular OS version).

It means that it is not possible to reliably identify BLE peripherals across devices based on their BLE IDs. Only a custom identification scheme implemented in peripheral's firmware can allow that. _A peripheral can also indicate whether it has bonding keys for a particular central device (smartphone)_, which may help avoiding unwanted bonding-request dialogs. Be aware that bonding keys can be deleted on the central device (unbind in system settings) and the peripheral wouldn't know that, so it's only a heuristic. The same is possible in the opposite direction: the peripheral can be reset, thus "forgetting" the bonding with a central device, while the central will assume it's bonded.

This is to say, that if you think you may benefit from querying the system for bonded devices, an even better cross-platform solution could be to extend your peripherals to facilitate your needs (rather than relying on a platform-specific functionality).