PhilipsHue/flutter_reactive_ble #228: subscribeToCharacteristic stream throws same error as the previous ConnectionUpdateState failure.

scottsidekick, 2021-02-17:
Here is the short story of what I am seeing on my device...

**Steps:**
-connect from flutter app to my custom hardware device.
-call subscribeToCharacteristic to listen to read characteristic.
-button on hardware device initiates a disconnection.
-connectToAdvertisingDevice stream posts a ConnectionStateUpdate with a state of disconnected and a GenericFailure with message: 
` Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 19 (GATT_CONN_TERMINATE_PEER_USER)`

Up to this point everything is fine. A ConnectStateUpdate came and communicated to me that a disconnection occurred, and some GenericFailure was present, but the connection state change was conveyed properly.

**Next Steps:**
-connect from flutter app to my custom hardware device.
-call subscribeToCharacteristic to listen to read characteristic.
-stream returned from subscribeToChacteristic throws an error (via onError callback) of the following type:
```
Type: GenericFailure<CharacteristicValueUpdateError>
Message: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 19 (GATT_CONN_TERMINATE_PEER_USER)
```
**Next Steps:**
-connect from flutter app to my custom hardware device.
-call subscribeToCharacteristic to listen to read characteristic.
-read characteristic connects correctly.

werediver, 2021-02-19:
You may be interested in reading through #223, in which a special case related to, but distinct from yours is being discussed.

The answer to your ticket is that this is the expected behavior. You requested a stream of updates from a peripheral and the stream was abruptly terminated together with the connection itself. Thus the stream communicates the reason for termination, which is the same error that caused the disconnection.

I hope that makes sense. Please check if this explanation fits your observations well and let us know.

remonh87, 2021-02-19:
Small addition the `GATT_CONN_TERMINATE_PEER_USER ` error means in general that the client on purpose terminated the connection because there was nothing to do. I would also check your peripheral.

werediver, 2021-02-19:
@remonh87 That is intentional:

> -button on hardware device initiates a disconnection.

scottsidekick, 2021-02-19:
Hi. I think you have misunderstood the issue here. The problem is not that the error is thrown from the device connection stream. The problem is that after the error is throw, and the ble library completely disconnect and cancel all stream, on a subsequent reconnect the same error is thrown **AGAIN** on the characteristic stream. In know way should this be the expected behaviour.

With this additional context, I think you should read through the initial post again completely and carefully.

To simplify even further...

-hardware device disconnects (expected)
-device connection stream send disconnect update with failure (expected)
-my code disconnects and cancels all streams (expected)
-scan and connect to the device again from the start
-after reconnecting to the device, the read characteristic redudantly throws the **same error again** (no expected).

It seems like there is some bad state that is not being cleared here.

werediver, 2021-02-19:
>  -after reconnecting to the device, the read characteristic redudantly throws the same error again (not expected).

Are you absolutely sure the problem here is not caused you your peripheral? Could you double-check?

The stream returned by method calls like `subscribeToCharacteristic` are constructed anew for each call and are based on the underlying multiplexed event channels (communication with the native side). There is very little state there.

If after all checks you still find the library behavior erroneous, put a delay _after_ you first disconnect everything and close and cancel everything and _before_ you initiate the connection again. Try a little (200–300 ms) and a bigger (1 s) delay, check, if either make any difference for you. This may give us some insight into the potential issue.

scottsidekick, 2021-02-19:
> Are you absolutely sure the problem here is not caused you your peripheral? Could you double-check?

I am not sure. 

I am not familiar with the firmware code of the device unfortunately. If you think it could be caused by the firmware I could look into it.

scottsidekick, 2021-02-19:
Additional info. Also symptomizes on Android 8.1.0.

remonh87, 2021-03-07:
@scottsidekick  did you have a chance to further look into it?

maciej-szulc, 2021-04-22:
Hey, I have the sam issue. Device Connected, then Disconnected, then after connecting device again, which goes well, subscribing to characteristics throws `Exception: GenericFailure<CharacteristicValueUpdateError>(code: CharacteristicValueUpdateError.unknown, message: "Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 8 (GATT_INSUF_AUTHORIZATION or GATT_CONN_TIMEOUT)")` even after long delay.

remonh87, 2021-04-27:
@maciej-szulc  the characteristic  that you are subscribing to has either: 

-  additional security. You need to  make sure that you first establish bonding with the peripheral.
-  The periperal  is not responsive at this time.. 
However these  errors are thrown by the android stack and there is  little  we can do against this .

closed 2021-04-27 by remonh87

wsmzdyss, 2022-09-08:
> Hi. There might be a misunderstanding here. The problem is not that the error is thrown from the device connection stream. The problem is that after the error is thrown, and the ble library completely disconnects and cancels all streams, on a subsequent reconnect the same error is thrown **AGAIN** on the characteristic stream.
> 
> To simplify even further...
> 
> -hardware device disconnects (expected) -device connection stream sends disconnect update with failure (expected) -my code disconnects and cancels all streams (expected) -scan and connect to the device again from the start -after reconnecting to the device, the read characteristic redundantly throws the **same error again** (not expected).
> 
> It seems like there is some bad state that is not being cleared here.

> Additional info. Also symptomizes on Android 8.1.0.

I have the same problem, did you finally solve it?